ARISE-Initiative/robomimic #11: Robosuite version check is not correctly working for v1.3

janghyuk-choi, 2021-11-23:
https://github.com/ARISE-Initiative/robomimic/blob/2804dd97dd1625ec861298a35cb677129d3bfacc/robomimic/envs/env_robosuite.py#L52

The comment says that "only support robosuite v0.3 and v1.2+". 
Since Robosuite v1.3 is released, the version check should be updated if Robotmimic actually supports v1.3.

amandlek, 2021-11-24:
Thank you for bringing this issue to our attention! Indeed, the repository is fully functional with robosuite datasets collected on v1.3 (but for reproducing our exact experiments we recommend being on the `offline_study` robosuite branch which is based off of v1.2) - we will fix this in an upcoming PR.

amandlek, 2021-12-20:
This should be fixed now (see [here](https://github.com/ARISE-Initiative/robomimic/blob/master/robomimic/envs/env_robosuite.py#L52)).

closed 2021-12-20 by amandlek