ARISE-Initiative/robomimic #15: using SequenceDataset as standalone module

Howuhh, 2021-12-20:
Hi! I want to use SequenceDataset in my project so I don't have to write it myself, but all other modules are not relevant for me.

When I try to create a dataset, I get an error:
```
AssertionError: error: must call ObsUtils.initialize_obs_utils_with_obs_config first
```
Can I somehow use a dataset without creating (or using minimal only relevant args) configs? For example, as in https://arise-initiative.github.io/robomimic-web/docs/introduction/examples.html, but without model creation and training loop.

Thanks!

Howuhh, 2021-12-20:
It this a correct way of achieving this if a want such obs as states?
```python
    obs_dict = {
        "low_dim": [
            "robot0_eef_pos",
            "robot0_eef_quat",
            "robot0_gripper_qpos",
            "object"
        ]
    }
    ObsUtils.initialize_obs_modality_mapping_from_dict(obs_dict)
    dataset = get_data_loader("path_to_data/lift/ph/low_dim.hdf5")
```

amandlek, 2021-12-20:
You can absolutely use the DataLoader on its own, but you might want to take some time to understand why the DataLoader cares about `ObsUtils` - the DataLoader needs to know which of your observations are images in order to do conversion from uint8 to float, re-scaling, and channel swap to prepare for training (see [here](https://github.com/ARISE-Initiative/robomimic/blob/master/robomimic/utils/dataset.py#L547)). 

If you will handle similar kinds of logic on your end, you can go ahead and remove all instances of this processing logic in the class, and then there won't be any dependence on `ObsUtils`.

closed 2021-12-20 by amandlek