ARISE-Initiative/robomimic #24: wrong rendering

Howuhh, 2022-02-21:
Hi! I want to visualize my agent's rollouts. However, the resulting videos turns out to be strange, I see some color artifacts. Should it be like this? 

What I do:
```python
# on each step
render_frames.append(env.render(mode="rgb_array", width=256, height=256))

# at the end
imageio.mimsave(render_path, render_frames, fps=32)
```

Result:
[image]

cremebrule, 2022-02-22:
Hi @Howuhh ,

Thanks for bringing up this issue. Are you using any custom logic beforehand (e.g.: wrappers)? The color artifacts you see are actually the collision meshes for those objects, which should be hidden by default.

Howuhh, 2022-02-22:
I have gym.ObservationWrapper which only concats state dim's in one vector (names are in obs_dict below), other methods are not overloaded. 

That's how I create env:
```python
def create_env(hdf5_path="data/lift/ph/low_dim.hdf5", render=False):
    obs_dict = {
        "low_dim": [
            "robot0_eef_pos",
            "robot0_eef_quat",
            "robot0_gripper_qpos",
            "object"
        ]
    }
    initialize_obs_modality_mapping_from_dict(obs_dict)

    env_meta = get_env_metadata_from_dataset(hdf5_path)
    env = create_env_from_metadata(
        env_meta=env_meta,
        render=render,
        render_offscreen=False,
        use_image_obs=False
    )
    return env
```

cremebrule, 2022-02-22:
Can you make sure `render_collision_mesh` is set to False when you're creating the environment? You can check by simply calling `env.render_collision_mesh`. (this is a kwarg passed into all environment constructors)

Howuhh, 2022-02-23:
Yes, they are definitely set to False, I even make an assert on that.

amandlek, 2023-07-04:
Closing this issue as it appears to be stale, and also more related to [robosuite](https://github.com/ARISE-Initiative/robosuite). Feel free to raise an issue there if problems persist.

closed 2023-07-04 by amandlek