ARISE-Initiative/robomimic #29: Goal-conditioned observations

PraveenElango, 2022-04-01:
Hello,

I noticed that there is a functionality to use goal-conditioned images in robomimic. I was interested to figure out how this worked and found out that we can use the `get_goal()` function in the `env_robosuite.py` file. However, this function was used only in the rollout runs and I couldn't find its use during the training. Is it possible to train using goal-conditioned observations in robomimic? I was thinking of instances such as goal-conditioned imitation learning where the image from the last time-step in the demonstration is used as the goal observation in the policy.

There was also a documentation comment in the `get_goal()` function noting that not all environments support this. I went to check all the environments in robosuite and saw that none of them had a `get_goal()` function. If I were to write my own `get_goal()` function in a robosuite environment, is it possible to return an agentview image in the function as a goal observation once the task has succeeded? Would appreciate any help on this, thank you!

danfeiX, 2022-04-01:
Re goal-conditioned training: you can set `goal_mode` in config to generate goal observation during training:
https://github.com/ARISE-Initiative/robomimic/blob/master/robomimic/utils/dataset.py#L433 

Re `get_goal()` in environment: it's still WIP to support this from the environment / robosuite side. A easy way to add goal is to read the last frame in a demonstration dataset and set the observation as the goal.

PraveenElango, 2022-04-02:
Thank you for getting back to me! 

However, I cannot use on-screen rendering when `use_camera_obs` is set to True whole collecting human demonstrations in robosuite. Hence, I have been collecting human demonstrations using the on-screen renderer and then extracting the image observations from the raw mujoco states as advised [here](https://github.com/ARISE-Initiative/robosuite/issues/311).

This means that in on-screen rendering mode, I will not be able to get the image observations in robosuite while collecting observations. Hence, is there a workaround for me to read the image observation in the last frame in a demonstration dataset and to set that observation as the goal?

PraveenElango, 2022-04-05:
Hello,

I would like to check if it is possible to train using goal-conditioning but disable goal-conditioning during rollouts (i.e. I don't provide a goal to the rollout policy). Thank you!

danfeiX, 2022-04-16:
i'm not sure what you mean by that. the model will need goal as input in order to make predictions.

amandlek, 2022-07-09:
Closing this issue - please re-open if problems persist.

closed 2022-07-09 by amandlek