ARISE-Initiative/robomimic #35: Download server still not working

zephirefaith, 2022-06-23:
Same as #34 still

amandlek, 2022-06-23:
Unfortunately the power outage is persisting - what specific datasets did you need? Perhaps we can find a different way to give it to you to unblock you, for now

zephirefaith, 2022-06-23:
Appreciate that @amandlek. I have some specific needs and want to introspect the data to see if I can extract the structure we need out of it. If I can get the Proficient-Human image-based dataset for Lift and Can for both sim and real, that'd be perfect. Does this data include depth images too or only RGB?

snasiriany, 2022-06-23:
Hi @zephirefaith we have Lift datasets in sim and real [in this temporary link](https://drive.google.com/drive/folders/1dQa2EY9wxIMFHm07wMMnmhsVR5BqQKni?usp=sharing).

Please note, after our dataset server is back up this link may expire. We hope this helps!

amandlek, 2022-07-09:
Servers have been back up for a while, so I am closing this issue.

closed 2022-07-09 by amandlek