ARISE-Initiative/robomimic #40: robosuite env.reset_to "hack" present in run_trained_agent.py but not in train_utils.py

MaxDu17, 2022-08-10:
I'm trying to use code from `run_trained_agent.py` to collect some rollout statistics and it seems like this particular script uses a `.reset_to()` call that resets an environment to its current state. To my knowledge, this trick isn't present in other robomimic evaluation scripts, like `run_rollout()` in `train_utils.py`. 

When collecting human demonstrations on the robosuite task, was the `.reset_to()` trick used? I'm seeing some performance differences between the two versions of eval scripts, and I'm trying to pinpoint the issue. 

https://github.com/ARISE-Initiative/robomimic/blob/b5d2aa9902825c6c652e3b08b19446d199b49590/robomimic/scripts/run_trained_agent.py#L103

amandlek, 2022-08-10:
Yup, you're right - we used the same trick during demo collection to ensure that the demonstrations could be deterministically played back on the same machine that was used for data collection. See [this](https://github.com/ARISE-Initiative/robosuite/blob/master/tests/test_environments/test_action_playback.py) for more information. What kinds of issues are you seeing?

MaxDu17, 2022-08-10:
Got it, ok! Yeah, so if I omit this reloading, a policy trained on the same data (which used this reloading during collection) has performed upwards of 10% worse. From your experience, what exactly does the reloading do in the environment? It almost seems like it shouldn't be doing anything because it's resetting to its own state, but that's clearly not the case.

amandlek, 2022-08-11:
I think it removes some numerical inaccuracies that might come from reading the state from the simulator and then setting it, but I'm not 100% on this. It's incredibly strange that you're seeing such a significant difference - have you been able to reproduce this error on one of our public datasets? It might also be worth cross-posting this issue on the robosuite repo so we can track it there as well.

MaxDu17, 2022-08-11:
Let me give this a try with one of the public datsets. I'll keep you posted!

MaxDu17, 2022-08-26:
I'll close this for now--I think it was a quirk of our own data, as I wasn't able to reproduce it on the public dataset. If something else comes up, I'll reopen it!

closed 2022-08-26 by MaxDu17