ARISE-Initiative/robomimic #41: ObsUtils.unprocess_obs_dict() modifies obs dict in-place

MaxDu17, 2022-08-11:
The `ObsUtils.unprocess_obs_dict()` seems to modify the observation dictionary that is passed in, in addition to returning it. For example, I observed that in the lines referenced below, `next_obs` images are between 0 and 1, while after line 147, `next_obs` images are between 0 and 255. This leads to a problem, as `obs` is derived from `next_obs`, which means that in the next loop around, we will pass already unprocessed images into `unprocess_obs_dict()`. This has led to some significant issues, as the saved images are corrupted from what is observed. As a simple fix, I wrapped the `next_obs` in `deepcopy(next_obs)` on line 147. 

https://github.com/ARISE-Initiative/robomimic/blob/b5d2aa9902825c6c652e3b08b19446d199b49590/robomimic/scripts/run_trained_agent.py#L142-L147

amandlek, 2022-08-11:
Great catch - thanks for reporting this bug! We will PR a fix soon.

MaxDu17, 2022-08-11:
Awesome! Thanks.

amandlek, 2022-08-11:
Would you be able to send over a trained model on one of our datasets so that we can reproduce this issue easily?

MaxDu17, 2022-08-11:
Does this work? https://drive.google.com/file/d/1QyN5YyZYhzJRAUcTPmRGmebkpyh_9-hy/view?usp=sharing 
It's from the can pick-place environment. I just ran the `run_trained_agent.py` script using this checkpoint from the repository, and the problem is there

amandlek, 2023-07-06:
Fixed in #71

closed 2023-07-06 by amandlek