ARISE-Initiative/robomimic #44: ERROR: GLEW initalization error: Missing GL version

datonefaridze, 2022-10-22:
Hi, I followed tutorials accoding to this [documentation](https://robomimic.github.io/docs/introduction/installation.html),
I have exported lines in my: .bashrc, .zshrc (i am using .zsh):
```
export LD_LIBRARY_PATH=/home/dato/.mujoco/mujoco210/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
export PATH="$LD_LIBRARY_PATH:$PATH"
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so
```
when i run: `python examples/train_bc_rnn.py --debug` to test my installation it gives me an error: `ERROR: GLEW initalization error: Missing GL version`
when i check my environmental variables in .zsh/bash terminal, `LD_PRELOAD` env variable is present, I don't know how to proceed forward

amandlek, 2022-10-22:
Can you give us more information on what platform and machine you're running on? To make sure we can try to reproduce your issue. This definitely seems like a [robosuite](https://github.com/ARISE-Initiative/robosuite) installation issue, so it might also be useful to raise an issue there.

datonefaridze, 2022-10-23:
@amandlek It's ubuntu 20.04, I think deleting the linke `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so` helped, but don't know why (I don't know exactly because when i rebooted it worked, so not sure which operation did the job(i tried many things), but it should be `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so`)

amandlek, 2022-10-23:
So I believe that this is due to the `mujoco-py` library - the `LD_PRELOAD` export needs to be there when using on-screen rendering, but it should be unset when running off-screen rendering (which is what we do during training runs to log videos)

closed 2022-10-23 by amandlek