ARISE-Initiative/robomimic #47: Online Environment

nuomizai, 2022-10-26:
Was it possible to access the online simulation environment?

amandlek, 2022-10-26:
Can you provide some more details on this issue? It's not clear what you mean

nuomizai, 2022-10-27:
For example, for task Lift/Can/Tool Hang/Square, you only provide the offline dataset robomimic v0.1. How can I  access the online environment to train my own agent or evaluate the policy trained with these offline datasets?

amandlek, 2022-10-27:
Our default training runs already evaluate the agent in the simulator (even though training is done offline) - see [here](https://robomimic.github.io/docs/introduction/getting_started.html#step-2-launch-experiment) for more information on default training configs. We also have [pretrained models](https://robomimic.github.io/docs/tutorials/using_pretrained_models.html) that you can evaluate in the simulator as well. Does this answer your question?

nuomizai, 2022-10-27:
> Can you provide some more details on this issue? It's not clear what you mean

For example, 

> pretrained models

Thanks for your reply. That addressed part of my concerns. I still would like to know if I can use robomimic for Deep Reinforcement Learning training, i.e., to get the standard observation and reward from the simulator and update the policy intermittently.

amandlek, 2022-10-27:
We don't support online RL at the moment, but I would recommend [this repo](https://github.com/ARISE-Initiative/robosuite-benchmark) for examples of how to run RL training with robosuite environments.

closed 2022-10-27 by amandlek