ARISE-Initiative/robomimic #49: Project Roadmap

lorepieri8, 2022-11-02:
Hi there, amazing project.

I'm considering building on top of this framework, but I would like to understand what are your plans for the future. I see that many libraries are outdated (mujoco-py not active, pytorch 1.6, python 3.7).

Which one of the below would you say is more accurate?
-Project will be maintaned with basic fixes and small updates.
-Project will be updated and will keep following the state of the art. 
-The project will be not be maintained anymore. 
-Other

amandlek, 2022-11-02:
Thanks for reaching out - the second option ("Project will be updated and will keep following the state of the art.") is most accurate. We have been working on a new version with some feature upgrades (stay tuned - this will also include support for the new python MuJoCo bindings from DeepMind), and are also happy to accept pull requests for useful features. I should also mention that the codebase is also compatible with later versions of pytorch and python.

closed 2022-11-02 by amandlek

lorepieri8, 2022-11-03:
Great! Can you share any ETA for the new version?

amandlek, 2022-11-04:
We don't have one that's concrete yet - a good estimate is the next month or two.

lorepieri8, 2023-04-20:
Was the update you mentioned part of the robosuite update or there is more to come for robomimic itself?

amandlek, 2023-04-21:
There is more to come for robomimic - we are updating our datasets to be compatible with the new robosuite update - this feature will be released in the next few weeks.