ARISE-Initiative/robomimic #50: ImportError: cannot import name 'postprocess_model_xml' from 'robosuite.utils.mjcf_utils'

supriyasathya, 2022-12-22:
I get this error when I run a coupe of scripts such as : python examples/train_bc_rnn.py --debug
 or even the playback_dataset.py script:

from robosuite.utils.mjcf_utils import postprocess_model_xml
ImportError: cannot import name 'postprocess_model_xml' from 'robosuite.utils.mjcf_utils' (/home/xyz/anaconda3/envs/robomimic_venv/lib/python3.7/site-packages/robosuite/utils/mjcf_utils.py)

Is 'postprocess_model_xml' removed/moved or renamed?

Facebear-ljx, 2022-12-24:
> I get this error when I run a coupe of scripts such as : python examples/train_bc_rnn.py --debug or even the playback_dataset.py script:
>
> from robosuite.utils.mjcf_utils import postprocess_model_xml [...]

I got the same error. It seems that the latest version of robosuite (1.4.0) removes 'postprocess_model_xml'. So, I tried using a version like 1.3.1 and it can avoid this error well.

Facebear-ljx, 2022-12-24:
> I get this error when I run a coupe of scripts such as : python examples/train_bc_rnn.py --debug or even the playback_dataset.py script:
>
> from robosuite.utils.mjcf_utils import postprocess_model_xml [...]

Using robosuite with version 1.2.1 works well. It seems that  'postprocess_model_xml' is removed in the latest robosuite (1.4.0).

amandlek, 2022-12-24:
Yes that's correct - we are in the process of updating robomimic to be compatible with v1.4 and will have a new release soon - stay tuned!

supriyasathya, 2022-12-31:
Ok! Thanks for the responses.

closed 2022-12-31 by supriyasathya

ZlSong417, 2023-07-17:
@amandlek @supriyasathya Hi!I have also encountered the same error in robosuite 1.4, how can I solve it?

amandlek, 2023-07-17:
Are you running the latest version of robomimic (e.g. v0.3)? If so, this should not be an issue anymore.

ZlSong417, 2023-08-08:
Yes, and then I switched to version 1.2 and it was fine.