ARISE-Initiative/robomimic #51: How to use transport environment ?

prajjwal1, 2023-01-06:
I get the following error when I pass in the path to transport data
`
Environment `TwoArmTransport` not found. Make sure it is a registered environment among: Lift, Stack, NutAssembly, .....
`
Does this error arise due to robosuite not having the `TwoArmTransport` environment ? Is yes, how do I reproduce the paper results on transport. Any suggestions will be helpful.

prajjwal1, 2023-01-06:
@amandlek I tried following all the instructions, seems like the current `robomimic` codebase doesn't work well with `offline_study` branch of `robosuite`. I get the following error, when I do `git checkout offline_study` and `pip install requirements.txt`.
```
Found 2 GPUs for rendering. Using device 0
ERROR: Shadow frame buffer is not complete. error 0x8cd7
```

amandlek, 2023-01-06:
This might be an issue with mujoco / `mujoco-py` and offscreen GPU rendering. Have you tried the environment out using robosuite on its own (for example `robosuite/demos/demo_random_action.py`)? Also if you could provide some more information on your exact setup and how to reproduce this issue, that'd be great.