ARISE-Initiative/robomimic #56: Problem about EGL when runnning train_bc_rnn.py

CHbloom, 2023-04-02:
When I was running python examples/train_bc_rnn.py --debug, all the code ran through, but an error was reported at the end：
Exception ignored in: <function MjRenderContext.__del__ at 0x7f2b77f2fef0>
Traceback (most recent call last):
  File "/DATACENTER1/jiangbo.chai/robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
    self.gl_ctx.free()
  File "/DATACENTER1/jiangbo.chai/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "/home/jiangbo.chai/anaconda3/envs/robomimic_venv/lib/python3.7/site-packages/OpenGL/error.py", line 234, in glCheckError
    baseOperation = baseOperation,
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglMakeCurrent,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f2b6dbe94d0>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f2b77fabb90>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f2b77fabb90>,
		<OpenGL._opaque.EGLContext_pointer object at 0x7f2b77fa5440>,
	),
	result = 0
)
Exception ignored in: <function EGLGLContext.__del__ at 0x7f2b77f2fe60>
Traceback (most recent call last):
  File "/DATACENTER1/jiangbo.chai/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
  File "/DATACENTER1/jiangbo.chai/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "/home/jiangbo.chai/anaconda3/envs/robomimic_venv/lib/python3.7/site-packages/OpenGL/error.py", line 234, in glCheckError
    baseOperation = baseOperation,
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglMakeCurrent,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f2b6dbe94d0>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f2b77fabb90>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f2b77fabb90>,
		<OpenGL._opaque.EGLContext_pointer object at 0x7f2b77fa5440>,
	),
	result = 0
)

Although this does not affect the operation of the code, I would like to ask what the problem is

amandlek, 2023-04-02:
Thanks for reaching out - I think this is related to the new mujoco DM bindings with EGL rendering - I have seen this error at the end of several of my scripts but it appears to be harmless, so we have not run any deeper investigation.

snasiriany, 2023-04-02:
Yes, I've also encountered this issue and it also has been harmless for me. If you want to avoid the issue, make sure to call `env.close()` on the underlying robosuite env instance.

closed 2023-04-03 by amandlek

CHbloom, 2023-04-15:
yes,I solve the problem by installing an older version of mujoco,and it runs without error

ojh6404, 2023-07-19:
@CHbloom  Im struggling with same error. what version of mujoco did you install?