ARISE-Initiative/robomimic #61: 404 error when trying to download expert data

trevorablett, 2023-06-29:
Hello,

Thanks for providing such an expansive set of environments/data. We tried to download some expert data this week, and were met with a 404 (at e.g., http://downloads.cs.stanford.edu/downloads/rt_benchmark/square/ph/low_dim.hdf5 and http://downloads.cs.stanford.edu/downloads/rt_benchmark/tool_hang/ph/low_dim.hdf5).

Is there an error on your end that's causing this? Thanks.

Trevor

trevorablett, 2023-06-29:
Oh that's good to hear! Waiting until Monday isn't a problem. Thanks for the prompt reply.

palashag12, 2023-07-01:
Hi I am running into the same issue and I really do need these datasets before monday. Is there any alternate way we can arrange  for recieveing these datasets?

amandlek, 2023-07-03:
The dataset links should be working now! Closing this issue.

closed 2023-07-03 by amandlek