ARISE-Initiative/robomimic #62: Velocity changes when collecting human demonstration data

dzy666fly, 2023-07-03:
Hi！Thank you so much for sharing the source code and datasets. Now I am confused of velocity changes when analyzing the Robomimc datasets. Considering that Lift is a simple task and the location of the cube is near the end-effector, it seems that higher velocity to locate the cube is more reasonable. So why do you choose a lower velocity to locate it? What's more, if we use our own data with BC or BC-RNN, the cube can be located precisely but cannot be graspped and picked up. Could you give us some advice?

For example, changes of the gripper's action on Robomimic datasets show that more time were spent in locating the cube and less time in grasping and picking up it. 
```
f1 = h5py.File('./datasets/robomimic/lift_real/ph/image.hdf5', 'r')
a1 = f1["data/demo_0/actions"][:,-1]

'''
print(a1)
[ 1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.
  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1.
  1.  1.  1.  1.  1.  1.  1.  1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1.
 -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1.]
print(len(a1), np.sum(a1==1.), np.sum(a1==-1.))
66 44 22
'''
```
Well, we try to use Spacemouse to collect data as you have done. However, we seem to find an opposed result; we spend less time in locating and more time in grasping and picking.
```
f2 = h5py.File('./dataset/own/lift_real/image.hdf5, 'r')
a2 = f2["data/demo_0/actions"][:,-1]

'''
52 11 41
[ 1.  1.  1.  1.  1.  1.  1.  1.  1.  1.  1. -1. -1. -1. -1. -1. -1. -1.
 -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1.
 -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1. -1.]
'''
```

amandlek, 2023-07-03:
I'm a bit unclear on what you're asking with respect to "velocity". However, different demonstrators might have different styles of teleoperation and use different strategies as well to solve different manipulation tasks. Indeed, this was one of the reasons we conducted our robomimic study. Differences such as trajectory length and amount of time spent on each part of the task are to be expected.

closed 2023-07-03 by amandlek

dzy666fly, 2023-07-03:
I'm so happy for your timely reply.
Emm, please allow me to describe my confusion in another way. When I try [https://github.com/ARISE-Initiative/robosuite/blob/offline_study/robosuite/demos/demo_device_control.py](url) with Spacemouse to finish the Lift task, the values of the action [https://github.com/ARISE-Initiative/robosuite/blob/offline_study/robosuite/demos/demo_device_control.py#L204](url) are shown as follows if we print them:
```
[0.007 0.000 -0.057 0.000 0.000 -0.000 -1.000] -1
[0.000 0.000 -0.062 0.000 0.000 -0.000 -1.000] -1
[0.005 0.000 -0.080 0.000 -0.003 -0.000 -1.000] -1
[0.020 0.000 -0.045 0.000 0.000 -0.000 -1.000] -1
[0.004 0.000 -0.034 0.000 0.000 -0.001 -1.000] -1
[0.000 0.000 -0.018 0.000 -0.001 -0.003 -1.000] -1
...
[-0.068 -0.002 -0.259 0.008 0.000 -0.000 -1.000] -1
[-0.068 0.000 -0.239 0.006 0.003 -0.000 -1.000] -1
[-0.061 0.000 -0.220 0.003 0.002 -0.000 -1.000] -1
[-0.046 0.000 -0.209 0.001 0.000 -0.000 -1.000] -1
[0.000 -0.009 -0.205 0.002 0.000 -0.000 1.000] 1
[0.000 -0.025 -0.188 0.002 0.000 -0.000 1.000] 1
[0.000 -0.013 -0.173 0.000 0.000 -0.000 1.000] 1
[0.000 -0.004 -0.145 0.000 0.000 -0.000 1.000] 1
...
[0.000 0.000 0.096 0.005 -0.009 -0.022 1.000] 1
[0.000 0.000 0.120 0.001 -0.004 -0.024 1.000] 1
[0.000 0.000 0.171 0.000 0.000 -0.026 1.000] 1
[0.000 0.000 0.184 0.000 -0.003 -0.026 1.000] 1
[0.000 0.000 0.177 0.001 0.000 -0.024 1.000] 1
[0.000 0.000 0.157 0.001 0.000 -0.024 1.000] 1
[0.000 0.000 0.123 0.003 0.000 -0.022 1.000] 1
[0.000 0.000 0.071 0.004 0.000 -0.018 1.000] 1
```
Obviously, the results show big difference from those in the Robomimic datasets as follows, nearly without the zero value in each row:
```
f1 = h5py.File('./datasets/robomimic/lift/ph/image.hdf5', 'r')
a1 = f1["data/demo_0/actions"][:]
print(a1)
'''
[[-0.00000000e+00  0.00000000e+00  0.00000000e+00  3.81497387e-03
   1.48207128e-01  1.44790206e-02 -1.00000000e+00]
 [ 2.04000000e-01  8.70000000e-02 -7.20000000e-02  4.07310342e-03
   1.37932464e-01  4.72844299e-03 -1.00000000e+00]
 [ 3.23000000e-01  1.31000000e-01 -7.30000000e-02  6.70890836e-03
   1.28519833e-01 -8.25768709e-03 -1.00000000e+00]
 [ 4.91000000e-01  2.10000000e-01 -1.90000000e-01  8.24865513e-03
   1.22598894e-01 -2.52794065e-02 -1.00000000e+00]
...
[-1.80000000e-01 -9.50000000e-02 -1.82000000e-01 -2.66734581e-03
  -4.57326658e-02  1.14844050e-02 -1.00000000e+00]
 [-1.87000000e-01 -1.20000000e-01 -1.60000000e-01 -2.09046574e-03
  -3.98743413e-02  1.51450215e-02 -1.00000000e+00]
 [-1.58000000e-01 -1.25000000e-01 -1.13000000e-01 -3.98083730e-03
  -3.54442075e-02  1.94974821e-02 -1.00000000e+00]
 [-1.80000000e-01 -1.11000000e-01 -1.17000000e-01  3.64121952e-04
  -2.56206505e-02  2.25369949e-02  1.00000000e+00]
 [-1.61000000e-01 -7.00000000e-03 -1.55000000e-01 -1.74624973e-03
  -1.73079297e-02  2.53038444e-02  1.00000000e+00]
...
 [ 1.30000000e-01 -1.62000000e-01  1.00000000e+00  1.32256988e-02
   8.09061751e-02  2.87604518e-02  1.00000000e+00]
 [ 1.91000000e-01 -1.61000000e-01  1.00000000e+00  1.33572202e-02
   1.05535321e-01  3.70519869e-02  1.00000000e+00]
 [ 2.29000000e-01 -1.31000000e-01  1.00000000e+00  1.13468701e-02
   1.26652688e-01  3.91908288e-02  1.00000000e+00]
 [ 2.64000000e-01 -9.50000000e-02  1.00000000e+00  6.96069142e-03
   1.41811416e-01  4.11663167e-02  1.00000000e+00]]
'''
```
So I'm confused of the way of teleoperation to get the similar actions as shown in the Robomimic datasets, which maybe improve the success rate of our own real world tasks.

amandlek, 2023-07-03:
The robomimic datasets were collected with a phone teleoperation interface, which may explain some of the differences. However, there have been many recent projects that have been successfully using the spacemouse interface for data collection - @snasiriany can comment further.