ARISE-Initiative/robomimic #79: Error about hdf5

warmmm, 2023-08-10:
I install robomimic today by the source, but when I try test the code , trhere is  the error :

AssertionError: @download_url got unreachable url: http://downloads.cs.stanford.edu/downloads/rt_benchmark/test_v141.hdf5

How to slove ?

amandlek, 2023-08-10:
This seems like a mujoco installation issue. We fixed an error where the code was still trying to import `mujoco_py` (see https://github.com/ARISE-Initiative/robomimic/commit/8df6e8e482458befaf1decfcd9d635d68a7afcae) - can you verify you're on the latest master commit, and that `mujoco-py` is not installed? Also, what version of the `mujoco` package are you on?

warmmm, 2023-08-11:
> 这似乎是 mujoco 安装问题。我们修复了代码仍在尝试导入的错误`mujoco_py`（请参阅[8df6e8e](https://github.com/ARISE-Initiative/robomimic/commit/8df6e8e482458befaf1decfcd9d635d68a7afcae)） - 您能否验证您是否使用最新的主提交，并且`mujoco-py`尚未安装？另外，`mujoco`你用的是什么版本的包？

mujoco version is 2.3.7  which is needed by the robosuite v1.4.1

amandlek, 2023-08-13:
Can you try a lower version of mujoco (maybe `mujoco==2.3.5`)? Also please verify that `mujoco-py` is not installed.

Nimingez, 2023-12-05:
Traceback (most recent call last):
  File "/home/nmz/robot/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
  File "/home/nmz/robot/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "/home/nmz/anaconda3/envs/robomimic/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglMakeCurrent,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f56ee800a40>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f57237ea840>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f57237ea840>,
		<OpenGL._opaque.EGLContext_pointer object at 0x7f57237ea7c0>,
	),
	result = 0
)

wangyan-hlab, 2023-12-12:
> Traceback (most recent call last): File "/home/nmz/robot/robosuite/robosuite/renderers/context/egl_context.py", line 155, in **del** self.free() File [...]

I met the same issue. Any solutions?

snasiriany, 2023-12-12:
I think this is an issue we can safely ignore. it happens when trying to free up the resources but is harmless

closed 2023-12-29 by amandlek