ARISE-Initiative/robomimic #85: 0 percent success rate on baseline method BC RNN with GMM

palashag12, 2023-09-05:
Hi
I have run the baseline BC_RNN_GMM on the master branch (without any changes) for square task for about 700 epochs with checkpoints every 50. I am consistently seeing a 0 percent success rate, which is clearly different from what is reported in the paper.

amandlek, 2023-09-05:
Please give us the exact steps you took so that we can reproduce this issue. This includes the specific branch / commit your are on for robosuite as well as robomimic, and the scripts (and config and dataset) you are using for training.

palashag12, 2023-09-05:
Hi, thanks for the timely response. The hash for the robosuite branch I am using is "4fef24619f65f031556db2c2f0b2840ca9cde7da"  , while the hash for the robomimic branch I am using is b5d2aa9902825c6c652e3b08b19446d199b49590" . I am using train.py with bc.json exactly as provided to me, only I make sure to enable both the "rnn" field and the gmm "field". The dataset I am using is proficient human for square task, 'low_dim.hdf5'  . Thank you

amandlek, 2023-09-05:
Can you follow the instructions [here](https://robomimic.github.io/docs/datasets/robomimic_v0.1.html#reproduce-study-results) to generate configs to reproduce our results? It is very likely that you are missing some crucial settings (for example, setting the config.train.seq_length to 10 for RNN training). You can then run a diff between the generated config and the one you are using to see the important settings you might be missing.

palashag12, 2023-09-26:
Hi
Thanks for the feedback, this worked!

closed 2023-09-26 by palashag12