ARISE-Initiative/robomimic #92: Questions about GMM based model architecture.

JisuHann, 2023-09-21:
Hello! I am trying to train [BC_Transformer_GMM](https://github.com/ARISE-Initiative/robomimic/blob/dca62a00127134437c28c96fe5629c5a9e3afda6/robomimic/algo/bc.py#L772) using the [VIOLA](https://github.com/UT-Austin-RPL/VIOLA) paper dataset, specifically for the stacking object task. The settings are identical to the VIOLA paper in both robosuite and robomimic. I have two questions:

1. **Gaussian Mixture Models (GMM) for Continuous Actions**: The object stacking task should predict the OSC_Position, where  four action parameters within the range of -1 to 1. I want to train a GMM to capture multi-modality. I'm curious if it's feasible to directly train GMM on continuous action parameters. Is there a need to discretize the actions, or can I directly compute the NLLoss (negative log-likelihood loss)? Additionally, I'd like to know the possible range of loss values for GMM (e.g., from infinity to 0). I once observed the loss starting at 10 and converging to a negative value like -8. I'm curious if such loss values are plausible.
2. **Timestep Consideration in TransformerGMMActorNetwork**: Inside the BC_Transformer_GMM, the [TransformerGMMActorNetwork](https://github.com/ARISE-Initiative/robomimic/blob/dca62a00127134437c28c96fe5629c5a9e3afda6/robomimic/models/policy_nets.py#L1131) returns a sequence of GMM distributions over the timesteps. The current dataset only provides the action distribution for the last timestep. If I want to make prediction for only the last timestep, should I additionally modify the code so that the model can only get the last timestep distribution and compute the loss, or do I need to compute the loss on the entire timestep distribution? I want to know whether there is a specific reason to output and compute the loss over the entire distribution sequence.

Thank you!

amandlek, 2023-09-21:
re (1): You do not need to discretize the actions. You can directly read our implementation of the GMM action prediction [here](https://github.com/ARISE-Initiative/robomimic/blob/40e427a695a9f085d3fc0f70ffb0d687a10802c8/robomimic/models/policy_nets.py#L494) and the loss is simply the negative log probability of the dataset actions (NLL). As for loss ranges, since probabilities are between 0 and 1, log probabilities should be strictly negative or 0, and NLL should always be non-negative, so it's a little strange that you are seeing negative loss values. 

re (2): I'm unfamiliar with the dataset you are using, but usually actions are provided at each timestep of a demonstration trajectory. Perhaps @zhuyifengzju can comment further?

closed 2023-09-21 by amandlek

zhuyifengzju, 2023-09-26:
For training, both VIOLA and robomimic implementations return distribution for each time step. The reason to output and compute the loss over the entire distribution sequence is based on the insight that predicting sequences can lead to better model performance. But I don't think you really need to stick to these designs if you find it confusing, and if you find performance doesn't differ much.