ARISE-Initiative/robomimic #98: Questions about dataset_states_to_obs.py

WangYixuan12, 2023-10-04:
Hi! Thank you all for the fantastic work! I am trying to look for one way to extract observations from the downloaded demo.hdf5. I found the [instruction](https://robomimic.github.io/docs/datasets/robosuite.html#extracting-observations-from-mujoco-states) in the tutorial. I liked the idea of it. However, I don't know how to extract depth observation from it. It seems that I can only extract RGB observations only. Do you have any ideas on how to extract depth observation from the demo.hdf5?

amandlek, 2023-10-04:
If you're talking about robosuite datasets, that would require a few changes to the script and the `EnvRobosuite` class. It could also require some other changes in the codebase to make sure depth images are processed correctly for training. We could put it on our roadmap for the next release if there's interest.

WangYixuan12, 2023-10-05:
Actually, I have found one way of rendering depth images. If you are interested, I could create a pull request and discuss it. If that has lower priority, we could close this issue.

snasiriany, 2023-10-05:
Please share the pull request with us!

WangYixuan12, 2023-10-05:
Thanks! I just created the pull request #99. We may discuss there.

closed 2023-10-05 by WangYixuan12