ARISE-Initiative/robomimic #101: Applying model to to the real robot

Soha18, 2023-10-06:
Hi,

Thanks for opening this repository.
I'm trying to implement lift task on real Panda robot after training a model on the simulator (robosuite).
even the success rate on the simulator is almost 100% but it didn't work in reality.
I'm using OSC controller and getting the observations from /franka_state_controller/franka_states and /joint_states topics.
I think the problem is that the coordinate system in the simulator and real world doesn't match. In the real robot I have the observations related to the base frame and couldn't be sure about the observations frame in the simulator.

any advice to solve this problem? did I miss something else??

Thanks a lot and I appreciate any help

danfeiX, 2023-10-20:
The repo is not designed for sim-to-real transfer. For all of our experiments in the original study paper, we collected demonstrations in the real world. For doing real-world data collection, you could try https://github.com/UT-Austin-RPL/deoxys_control

closed 2023-10-20 by amandlek

Nimingez, 2023-12-06:
how about ur5?