ARISE-Initiative/robomimic #105: 'EnvRobosuite' object has no attribute 'use_depth_obs'

praj441, 2023-10-19:
robomimic/robomimic/envs/env_robosuite.py", line 187, in render
    if self.use_depth_obs:
AttributeError: 'EnvRobosuite' object has no attribute 'use_depth_obs'

I am using Ubuntu 20.04 with python 3.8
Robosuite version 1.4.1

amandlek, 2023-10-19:
Great catch! https://github.com/ARISE-Initiative/robomimic/pull/106 should fix this (just pending approval before merging)

praj441, 2023-10-19:
Okay. That's great.
After this resolved I am getting another error:

robomimic/robomimic/envs/env_robosuite.py", line 62, in __init__
    self._is_v1 = (robosuite.__version__.split(".")[0] == "1")
AttributeError: module 'robosuite' has no attribute '__version__'

amandlek, 2023-10-19:
That's strange. What happens if you run `import robosuite` and then print `robosuite.__version__` in a terminal?

praj441, 2023-10-19:
solved the above error (__version__).
The error is coming due to the robosuite installed from source (I don't know , why).
I uninstalled robosuite and installed again using pip command directly.
And it worked !

mnm117, 2023-10-19:
the error exists on the Colab sample which the team provided. The code runs with Robosuite installation from pip.

amandlek, 2023-10-20:
Closing this issue as it is now fixed on the master branch. For any problems with the colab notebook, please open a new issue. Thanks!

closed 2023-10-20 by amandlek

amandlek, 2023-10-20:
I believe @danfeiX has fixed the issue with the colab notebook with this PR: https://github.com/ARISE-Initiative/robomimic/pull/108