ARISE-Initiative/robomimic #107: RNN Sequence length training (does it overlap or not?)

palashag12, 2023-10-20:
Hi
When a sequence length is picked for the RNN training process, does it overlap over the time series data for the demonstration or not?
For example, if seq length is 10, will 1 to 10 be the first sequence and timesteps 2 to 11 be used for the next training step, or will the next training step be timesteps 10 to 20?
Thanks

danfeiX, 2023-10-20:
they are overlapping sequences

closed 2023-10-20 by amandlek

palashag12, 2023-10-23:
Hello
I would like to add a follow up question to this. If the total sequence is 1 -100, will the final batch sequence be one that is 91-100. And if this is the case, let's say I have the RNN with a sequence length but I also have a second network for my method that I want to train simultaneously as both are needed during execution. This second method should only be conditioned on the current observation. Is there an easy way of incorporating this in the code so that with a sequence length of 100, I can still include all observations and condition the network only on the current observation.
Thanks

amandlek, 2023-10-23:
We actually "pad" the end of the sequence by default - see the `pad_seq_length` parameter explained on [this page](https://robomimic.github.io/docs/modules/dataset.html), or feel free to check out the source code in `robomimic/utils/dataset.py`.

As for using the same batch with different networks, you could just index out the time dimension for the network that needs only the current observation.