ARISE-Initiative/robomimic #114: Installation Issue

Dhanushvarma, 2023-11-12:
The following error keeps occuring when running
`pip install -e .`

`  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for egl_probe
  Running setup.py clean for egl_probe
Failed to build egl_probe
ERROR: Could not build wheels for egl_probe, which is required to install pyproject.toml-based projects
`

and this error even occurs when trying to do `pip install egl_probe` . How do i fix this, also are there any independent scripts available for postprocessing the hd5 file collected in "Robosuite" and to extract image observation from it? 

Thanks

snasiriany, 2023-11-13:
what if you tried `pip install egl-probe==1.0.2`?

Dhanushvarma, 2023-11-15:
Still the same issue.

closed 2023-11-19 by Dhanushvarma

theonlynick0430, 2024-01-08:
This error has occured for me as well

CallMeFrozenBanana, 2024-12-20:
`sudo apt install build-essential` can solve my problem.
ref: https://github.com/huggingface/lerobot/issues/105#issuecomment-2096007109

haoyizheng2000, 2025-06-16:
cmake version matters. I also met this issue. For me, `pip install cmake ` and `sudo apt install build-essential` did not work. I solved it with `pip install cmake==3.22`, as suggested [https://github.com/isaac-sim/IsaacLab/issues/1514](url).

lorepieri8, 2025-07-04:
cmake 3.24.2 worked for me on Mac, to avoid this issue: https://stackoverflow.com/questions/74194672/cmake-problems-after-upgrading-to-macos-13-0