ARISE-Initiative/robomimic #116: get_dataset_info for demos collect from Robosuite

Dhanushvarma, 2023-11-23:
After collecting demonstrations in the robosuite env, and then converting them using the conversion script provided, there seems to be an issue of action not being in the scale [-1, 1] . Can this issue be ignored?

`==== Dataset Structure ====
episode demo_1 with 482 transitions
    key: actions with shape (482, 7)
    key: states with shape (482, 32)

Traceback (most recent call last):
  File "get_dataset_info.py", line 134, in <module>
    raise Exception("Dataset should have actions in [-1., 1.] but got bounds [{}, {}]".format(action_min, action_max))
Exception: Dataset should have actions in [-1., 1.] but got bounds [-3.7500000000000036, 3.7500000000000036]`

amandlek, 2023-11-28:
That's unexpected... @snasiriany have you also encountered this when collecting demonstrations with robosuite and spacemouse?

snasiriany, 2023-11-28:
Using collect_human_demonstrations.py to collect demos, I've never seen this issue. I have seen it in other contexts with my own custom scripts. @Dhanushvarma how exactly did you obtain this dataset? Did you collect it yourself, using collect_human_demonstrations.py?

Dhanushvarma, 2023-11-28:
@snasiriany Yes, I have collected the demonstrations using "collect_human_demonstrations.py". However, one thing to note is that, I was collecting demonstrations on a slightly modified "Lift" task, with there being 2 blocks to choose from, not sure if that's relevant to the issue above.

snasiriany, 2023-11-28:
I'd recommend checking the actions are within bounds [-1, 1] while you're collecting the demonstrations to pin down the issue. Let us know what you find.

Nimingez, 2023-12-08:
I have collected the demonstrations using "collect_human_demonstrations.py". 
total transitions: 27087
total trajectories: 35
traj length mean: 773.9142857142857
traj length std: 240.8114106727588
traj length min: 362
traj length max: 1490
action min: -7.500000000000007
action max: 7.5

==== Filter Keys ====
filter key train with 32 demos
filter key valid with 3 demos

==== Env Meta ====
{
    "type": 1,
    "env_name": "Wipe",
    "env_version": "1.4.1",
    "env_kwargs": {
        "env_name": "Wipe",
        "robots": "UR5e",
        "controller_configs": {
            "type": "OSC_POSE",
            "input_max": 1,
            "input_min": -1,
            "output_max": [
                0.05,
                0.05,
                0.05,
                0.5,
                0.5,
                0.5
            ],
            "output_min": [
                -0.05,
                -0.05,
                -0.05,
                -0.5,
                -0.5,
                -0.5
            ],
            "kp": 150,
            "damping_ratio": 1,
            "impedance_mode": "fixed",
            "kp_limits": [
                0,
                300
            ],
            "damping_ratio_limits": [
                0,
                10
            ],
            "position_limits": null,
            "orientation_limits": null,
            "uncouple_pos_ori": true,
            "control_delta": true,
            "interpolation": null,
            "ramp_ratio": 0.2
        }
    }
}

==== Dataset Structure ====
episode demo_1 with 1129 transitions
    key: actions with shape (1129, 6)
    key: states with shape (1129, 13)

Traceback (most recent call last):
  File "get_dataset_info.py", line 134, in <module>
    raise Exception("Dataset should have actions in [-1., 1.] but got bounds [{}, {}]".format(action_min, action_max))
Exception: Dataset should have actions in [-1., 1.] but got bounds [-7.500000000000007, 7.5]

Nimingez, 2023-12-08:
your test dataset 
[image]
my dataset
[image]

Nimingez, 2023-12-11:
@Dhanushvarma Have you solved this problem?

Nimingez, 2023-12-11:
How did you gather this information robosuite/robosuite/models/assets/demonstrations/wipe/panda

Dhanushvarma, 2023-12-12:
> @Dhanushvarma Have you solved this problem?

No, the issue still persists

Dhanushvarma, 2023-12-12:
> I'd recommend checking the actions are within bounds [-1, 1] while you're collecting the demonstrations to pin down the issue. Let us know what you find.

When using the keyboard controller for collecting the demonstrations, are the actions not scaled down automatically?

Dhanushvarma, 2023-12-13:
> I'd recommend checking the actions are within bounds [-1, 1] while you're collecting the demonstrations to pin down the issue. Let us know what you find.

I have printed the actions in the script and they are out of bounds with 3.75 being the magnitude, don't know why this number/scale is appearing

amandlek, 2023-12-13:
This sounds like an issue with robosuite and the keyboard demonstration collection interface. Can you raise this issue on that repository so the robosuite team can look into it further? Thank you!

closed 2023-12-13 by amandlek