ARISE-Initiative/robomimic #123: Cannot download the robomimic dataset with image

StarCycle, 2023-12-20:
Hello, 

I cannot download the robomimic dataset with the following url:

http://downloads.cs.stanford.edu/downloads/rt_benchmark/lift/mg/image_sparse_v141.hdf5
http://downloads.cs.stanford.edu/downloads/rt_benchmark/lift/mg/image_dense_v141.hdf5
http://downloads.cs.stanford.edu/downloads/rt_benchmark/can/mg/image_sparse_v141.hdf5
http://downloads.cs.stanford.edu/downloads/rt_benchmark/can/mg/image_dense_v141.hdf5

I get these url from the DATASET_REGISTRY dict in your colab script (https://colab.research.google.com/drive/1b62r_km9pP40fKF0cBdpdTO2P_2eIbC6?usp=sharing#scrollTo=cfsnmbDYv7xZ) 

Do you change the address of the dataset? Or there is a newer version?

I am looking for a dataset with images in simulated environment, so I can evaluate the policy in your simulator without sim2real.

Kind regards,
Zhuoheng

amandlek, 2023-12-26:
Thanks for pointing this out! We don't actually have direct download links for these datasets that are based on robosuite v1.4.1 (see note [here](https://robomimic.github.io/docs/datasets/robomimic_v0.1.html#downloading) about this change). Instead, we recommend that you extract image observations from the `demo_v141.hdf5` datasets. See [here](https://robomimic.github.io/docs/datasets/robosuite.html#extracting-observations-from-mujoco-states) for more info on this, and [here](https://github.com/ARISE-Initiative/robomimic/blob/bdc89f7a1975aa3df46c5a9656afd22517e999c7/robomimic/scripts/extract_obs_from_raw_datasets.sh#L20) for example commands you can use. 

We will change these links in the DATASET_REGISTRY to avoid confusion (tagging @snasiriany to handle this).

snasiriany, 2023-12-28:
I started a PR to fix this confusion: https://github.com/ARISE-Initiative/robomimic/pull/127

closed 2023-12-29 by amandlek