ARISE-Initiative/robomimic #126: Loading transport dataset

hari-sikchi, 2023-12-26:
Hi,

I am trying to setup the transport task in the paper: I am on the main branch of robomimic and offline_study branch of robosuite.

The dataset url link seems to be missing:
` python download_datasets.py --tasks transport --dataset_types ph --hdf5_types image --download_dir /datastor1/hsikchi/robomimic_datasets/

Downloading dataset:
    task: transport
    dataset type: ph
    hdf5 type: image
    download path: /datastor1/hsikchi/robomimic_datasets/transport/ph
Skipping transport-ph-image, no url for dataset exists. Create this dataset locally by running the appropriate command from robomimic/scripts/extract_obs_from_raw_datasets`

When I am trying to make the dataset locally, I am getting the following error:

`Traceback (most recent call last):                                                                                                                                                                                                                                   [655/1945]
  File "dataset_states_to_obs.py", line 425, in <module>
    dataset_states_to_obs(args)
  File "dataset_states_to_obs.py", line 260, in dataset_states_to_obs
    traj, camera_info = extract_trajectory(
  File "dataset_states_to_obs.py", line 92, in extract_trajectory
    obs = env.reset_to(initial_state)
  File "/scratch/cluster/hsikchi/work/robomimic/robomimic/envs/env_robosuite.py", line 158, in reset_to
    self.env.reset_from_xml_string(xml)
  File "/scratch/cluster/hsikchi/work/robosuite/robosuite/environments/base.py", line 501, in reset_from_xml_string
    self._initialize_sim(xml_string=xml_string)
  File "/scratch/cluster/hsikchi/work/robosuite/robosuite/environments/base.py", line 216, in _initialize_sim
    self.mjpy_model = load_model_from_xml(xml_string) if xml_string else self.model.get_model(mode="mujoco_py")
  File "cymj.pyx", line 191, in mujoco_py.cymj.load_model_from_xml
Exception: <?xml version='1.0' encoding='utf8'?>
<mujoco model="base">
  <compiler angle="radian" meshdir="meshes/" autolimits="true" />
  <option impratio="20" density="1.2" viscosity="2e-05" cone="elliptic" />
  <size njmax="5000" nconmax="5000" />
  <visual>
    <map znear="0.001" />
  </visual>
  <default />
  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.9 0.9 1" rgb2="0.2 0.3 0.4" width="256" height="1536" />
    <texture type="2d" name="texplane" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/arenas/../textures/light-gray-floor-tile.png" />
    <texture type="cube" name="tex-ceramic" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/arenas/../textures/ceramic.png" />
    <texture type="cube" name="tex-steel-brushed" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/arenas/../textures/steel-brushed.png" />
    <texture type="2d" name="tex-light-gray-plaster" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/arenas/../textures/light-gray-plaster.png" />
    <texture type="cube" name="textable" builtin="flat" rgb1="0.5 0.5 0.5" width="512" height="3072" />
    <texture type="cube" name="payload_metal" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/steel-scratched.png" />
    <texture type="cube" name="payload_wood" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/light-wood.png" />
    <texture type="cube" name="trash_redwood" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/red-wood.png" />
    <texture type="cube" name="transport_start_bin_dark_wood" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/dark-wood.png" />
    <texture type="cube" name="transport_target_bin_dark_wood" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/dark-wood.png" />
    <texture type="cube" name="transport_trash_bin_dark_wood" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/dark-wood.png" />
    <texture type="cube" name="transport_start_bin_lid_dark_wood" file="/scratch/cluster/hsikchi/work/robosuite/robosuite/models/assets/textures/dark-wood.png" />
    <material name="floorplane" texture="texplane" texuniform="true" texrepeat="2 2" specular="0" shininess="0" reflectance="0.01" />
    <material name="table_ceramic" texture="tex-ceramic" specular="0.2" shininess="0" />
    <material name="table_legs_metal" texture="tex-steel-brushed" shininess="0.8" reflectance="0.8" />
    <material name="walls_mat" texture="tex-light-gray-plaster" texuniform="true" texrepeat="3 3" specular="0.1" shininess="0.1" />
    <material name="table_mat" texture="textable" />
    <material name="robot0_Face636_001" shininess="0.45" rgba="0.901961 0.921569 0.929412 1" />
    <material name="robot0_Part__Feature017_001" shininess="0.45" />
    <material name="robot0_Part__Feature018_001" shininess="0.45" />
    <material name="robot0_Part__Feature019_001" shininess="0.45" />
    <material name="robot0_Part__Feature022_001" shininess="0.45" rgba="0.901961 0.921569 0.929412 1" />
    <material name="robot0_Part__Feature023_001" shininess="0.45" rgba="0.25098 0.25098 0.25098 1" />
....
Failed to load XML from string. mj_loadXML error: b"XML Error: Schema violation: unrecognized attribute: 'autolimits'\n\nElement 'compiler', line 3, column 3\n"
  
    `

amandlek, 2023-12-26:
I believe this error is coming from `mujoco_py` which is deprecated when using the latest version of these datasets (see the note [here](https://robomimic.github.io/docs/datasets/robomimic_v0.1.html#downloading)). Instead, we recommend using robosuite v1.4.1 along with the DeepMind MuJoCo bindings (which can be installed with a simple `pip install mujoco`).

closed 2023-12-26 by amandlek

hari-sikchi, 2023-12-27:
Using the robosuite v1.4.1 I run into a different error:

this is the script to extract images from the raw data file in robomimic:
https://github.com/ARISE-Initiative/robomimic/blob/bdc89f7a1975aa3df46c5a9656afd22517e999c7/robomimic/scripts/extract_obs_from_raw_datasets.sh#L83
but executing this gives the following error:
[image]
it seems this error is because we are asking images from robot0 and robot1 whereas the dataset script has a hard check https://github.com/ARISE-Initiative/robomimic/blob/bdc89f7a1975aa3df46c5a9656afd22517e999c7/robomimic/scripts/dataset_states_to_obs.py#L189

amandlek, 2023-12-27:
Thanks for catching this! I've created a new branch that should fix this issue (https://github.com/ARISE-Initiative/robomimic/tree/fix_obs_extract_eye_in_hand). Can you please give it a try and let me know that it's working for you? Then we can merge it into master.

hari-sikchi, 2023-12-28:
yes this works