ARISE-Initiative/robomimic #131: Cannot generate agentview and eye_in_hand_view at the same time

StarCycle, 2024-01-17:
Hello, 

I am new to robomimic and I find the following problems:

1. I cannot generate agentview and eye_in_hand_view at the same time. I use the following code

```
env_meta = FileUtils.get_env_metadata_from_dataset("image_sparse_v141.hdf5")
config = config_factory(algo_name="bc")
ObsUtils.initialize_obs_utils_with_config(config)
env = EnvUtils.create_env_from_metadata(
        env_meta=env_meta,
        env_name=env_meta["env_name"],
        render=False,
        render_offscreen=True,
        use_image_obs=True,
)
```
I get the dataset according to issue #123 with this command:
```
python dataset_states_to_obs.py --done_mode 0 \
--dataset /content/robomimic_data/demo_v141.hdf5 \
--output_name image_sparse_v141.hdf5 --camera_names agentview robot0_eye_in_hand --camera_height 84 --camera_width 84
```
The env_meta['env_kwargs']['camera_names'] is
`['agentview', 'robot0_eye_in_hand']`. However, when I call `obs=env.reset()`, `obs` only contains `robot0_eye_in_hand_image`. There is no `agentview_image`.

2. I want to use a Gym version of 'Lift' environment, so I can run multiple simulators in parallel with `gym.vector.AsyncVectorEnv`. How to use the Gym wrapper? If I simply set `env_meta['type']=2`, I get the following error:
```
WARNING: could not load d4rl environments!
gym.error.NameNotFound: Environment `Lift` doesn't exist.
```

3. Since `get_env_metadata_from_dataset` wraps a robosuite environment, I can directly create a robosuite environment to solve the problems mentioned above.  However, When I create a 'Lift' environment using robosuite with
```
env = suite.make(
        env_name="Lift",
        robots="Panda",
        has_renderer=False,
        has_offscreen_renderer=True,
        use_camera_obs=True,
        camera_heights = 84,
        camera_widths = 84,
        camera_names = ['agentview', 'robot0_eye_in_hand'],
)
```
An action in such environment will have 8 dimensions (instead of 7 dimensions in the robomimic dataset)...

amandlek, 2024-02-01:
For your first question, I'm guessing it is because you're using `config = config_factory(algo_name="bc")` to initialize ObsUtils, but the default config will not have image observations enabled. Instead, take a look at this snippet [here](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/envs/env_robosuite.py#L481), which shows an example of informing ObsUtils of which observation modalities are images. This is because the environment checks to include rgb observations [here](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/envs/env_robosuite.py#L211). 

For your second question, robosuite has a [GymWrapper](https://github.com/ARISE-Initiative/robosuite/blob/b9d8d3de5e3dfd1724f4a0e6555246c460407daa/robosuite/wrappers/gym_wrapper.py) but we have not built a robomimic wrapper for this. You might need some custom implementation to support this use case.

For your third question, this is because robosuite expects a controller config when initializing the environment. I believe the default controller is a joint_velocity controller (which has an 8-dim action space) while we use the OSC_POSE controller for our experiments (which is 7-dim).

closed 2024-02-01 by amandlek

StarCycle, 2024-02-14:
Hello, since the data is collected with the OSC_POSE controller, the "actions" field of robomimic dataset can only be in EE action space. I cannot train the policy with joint position actions. Is my understanding correct?

amandlek, 2024-02-14:
This is correct.