ARISE-Initiative/robomimic #132: Problem with visualizing the dataset

AaronChuh, 2024-01-28:
Hi, when I try to visualize the dataset `threading_d1_panda.hdf5` downloaded in huggingface with command `python robomimic/scripts/playback_dataset.py --dataset threading_d1_panda.hdf5 --render_image_names agentview robot0_eye_in_hand --video_path /tmp/playback_dataset.mp4 --n 5`, I got the error below:

> ~/code/robomimic$ python robomimic/scripts/playback_dataset.py --dataset threading_d1_panda.hdf5 --render_image_names agentview robot0_eye_in_hand --video_path /tmp/playback_dataset.mp4 --n 5
> 
> ============= Initialized Observation Utils with Obs Spec =============
> 
> using obs modality: low_dim with keys: ['robot0_eef_pos']
> using obs modality: rgb with keys: []
> [robosuite WARNING] No private macro file found! (macros.py:53)
> [robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
> [robosuite WARNING] To setup, run: python /home/aaron/code/robosuite/robosuite/scripts/setup_macros.py (macros.py:55)
> Traceback (most recent call last):
>   File "/home/aaron/code/robomimic/robomimic/scripts/playback_dataset.py", line 392, in <module>
>     playback_dataset(args)
>   File "/home/aaron/code/robomimic/robomimic/scripts/playback_dataset.py", line 239, in playback_dataset
>     env = EnvUtils.create_env_from_metadata(env_meta=env_meta, render=args.render, render_offscreen=write_video)
>   File "/home/aaron/code/robomimic/robomimic/utils/env_utils.py", line 229, in create_env_from_metadata
>     env = create_env(
>   File "/home/aaron/code/robomimic/robomimic/utils/env_utils.py", line 174, in create_env
>     env = env_class(
>   File "/home/aaron/code/robomimic/robomimic/envs/env_robosuite.py", line 100, in __init__
>     self.env = robosuite.make(self._env_name, **kwargs)
>   File "/home/aaron/code/robosuite/robosuite/environments/base.py", line 35, in make
>     raise Exception(
> Exception: Environment Threading_D1 not found. Make sure it is a registered environment among: Lift, Stack, NutAssembly, NutAssemblySingle, NutAssemblySquare, NutAssemblyRound, PickPlace, PickPlaceSingle, PickPlaceMilk, PickPlaceBread, PickPlaceCereal, PickPlaceCan, Door, Wipe, ToolHang, TwoArmLift, TwoArmPegInHole, TwoArmHandover, TwoArmTransport

However, when I try the dataset `low_dim_v141.hdf5` downloaded with script, I got the new error:

> ~/code/robomimic$ python robomimic/scripts/playback_dataset.py --dataset /home/aaron/code/robomimic/datasets/lift/ph/low_dim_v141.hdf5 --render_image_names agentview robot0_eye_in_hand --video_path /tmp/playback_dataset.mp4 --n 5
> 
> ============= Initialized Observation Utils with Obs Spec =============
> 
> using obs modality: low_dim with keys: ['robot0_eef_pos']
> using obs modality: rgb with keys: []
> [robosuite WARNING] No private macro file found! (macros.py:53)
> [robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
> [robosuite WARNING] To setup, run: python /home/aaron/code/robosuite/robosuite/scripts/setup_macros.py (macros.py:55)
> Created environment with name Lift
> Action size is 7
> Playing back episode: demo_0
> Playing back episode: demo_1
> Playing back episode: demo_2
> Playing back episode: demo_3
> Playing back episode: demo_4
> Exception ignored in: <function MjRenderContext.__del__ at 0x7f3ec14d8040>
> Traceback (most recent call last):
>   File "/home/aaron/code/robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
>     self.gl_ctx.free()
>   File "/home/aaron/code/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
>     EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
>   File "/home/aaron/anaconda3/envs/RoboEnv/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError
>     raise self._errorClass(
> OpenGL.raw.EGL._errors.EGLError: EGLError(
> 	err = EGL_NOT_INITIALIZED,
> 	baseOperation = eglMakeCurrent,
> 	cArguments = (
> 		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f3e8428a0c0>,
> 		<OpenGL._opaque.EGLSurface_pointer object at 0x7f3ec16545c0>,
> 		<OpenGL._opaque.EGLSurface_pointer object at 0x7f3ec16545c0>,
> 		<OpenGL._opaque.EGLContext_pointer object at 0x7f3ec1654540>,
> 	),
> 	result = 0
> )
> Exception ignored in: <function EGLGLContext.__del__ at 0x7f3ec14d4e50>
> Traceback (most recent call last):
>   File "/home/aaron/code/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
>     self.free()
>   File "/home/aaron/code/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
>     EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
>   File "/home/aaron/anaconda3/envs/RoboEnv/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError
>     raise self._errorClass(
> OpenGL.raw.EGL._errors.EGLError: EGLError(
> 	err = EGL_NOT_INITIALIZED,
> 	baseOperation = eglMakeCurrent,
> 	cArguments = (
> 		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f3e8428a0c0>,
> 		<OpenGL._opaque.EGLSurface_pointer object at 0x7f3ec16545c0>,
> 		<OpenGL._opaque.EGLSurface_pointer object at 0x7f3ec16545c0>,
> 		<OpenGL._opaque.EGLContext_pointer object at 0x7f3ec1654540>,
> 	),
> 	result = 0
> )

Could you tell me how to fix the problem? Thanks!!!

amandlek, 2024-01-29:
For the first error, you need [this import](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/envs/env_robosuite.py#L14) to go through - I'm guessing something is going wrong here. 

For the second error, you shouldn't need this as long as you are on the latest robosuite version, but can you try prefacing your command with `MUJOCO_GL=egl` and see if that fixes your problem?

AaronChuh, 2024-01-29:
> For the first error, you need [this import](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/envs/env_robosuite.py#L14) to go through - I'm [...]

Thanks!!! I fix the first error by installing mimicgen, and for the second error, I found it's due to the incompatibility between the versions of GPU driver and CUDA. And thanks for your nice work!

closed 2024-01-29 by AaronChuh