ARISE-Initiative/robomimic #138: How to extract robot state information from demo.hdf5?

Vhyrsd, 2024-02-26:
@amandlek We used `python dataset_states_to_obs.py --dataset /path/to/demo.hdf5 --output_name low_dim.hdf5 --done_mode 2` to extract the observation data from the demo.hdf5 file. However, low_dim.hdf5 only contains the low dim data "object". How can we obtain the robot state information like "robot0_eef_pos" and "robot0_joint_pos"? Thank you so much!

Vhyrsd, 2024-02-27:
@amandlek Thank you for your reply. 
We first used `python collect_human_demonstrations.py --environment Lift --directory /home/smart/robosuite-1.2/robosuite/human_data/0 --device spacemouse` in robosuite 1.2 to collect the human demonstrations data file demo.hdf5.
Then we used `python convert_robosuite.py --dataset /home/smart/robosuite-1.2/robosuite/human_data/0/1708923255_2394037/demo.hdf5` in robomimic to convert the demo.hdf5.
Finally we use `python dataset_states_to_obs.py --dataset /home/smart/robosuite-1.2/robosuite/human_data/0/1708923255_2394037/demo.hdf5 --output_name image_14.hdf5 --done_mode 2 --camera_names agentview robot0_eye_in_hand --camera_height 84 --camera_width 84` to obtain the state information. It is worth noting that we found the "low_dim with keys" was an empty list.
[image]
Is it possible to setup the keys of low dim observations somewhere? Thank you so much!

amandlek, 2024-02-27:
The low_dim keys are expected to be empty (see [here](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/envs/env_robosuite.py#L475)).

Which exact branch and commit of robosuite are you on? If you want to use v1.2, you should try the [offline_study](https://github.com/ARISE-Initiative/robosuite/tree/offline_study) branch out. I suspect that you robosuite version is too old, and that might be causing the problem. 

One other thing worth trying is to try observation extraction on one of our existing datasets instead of one you collected yourself.

Vhyrsd, 2024-02-27:
@amandlek Thank you for your reply. We try to use the newest version of robosuite (1.4). However, we find that the program "collect_human_demonstrations.py" cannot quit by pressing the "Esc" key when either using keyboard or spacemouse, which makes us can not record the data correctly.

amandlek, 2024-02-28:
Have you tried running observation extraction on one of our existing datasets? And did you also try using the  `offline_study` branch of robosuite?

closed 2024-07-01 by amandlek