ARISE-Initiative/robomimic #139: Extracting low dimensional observations

Dhanushvarma, 2024-02-28:
In robosuite, for the environment, I had given some initialization noise (dictionary of type and magnitude) which was default set to zero. I believe this applies noise onto the starting joint angles of the robot. With this configuration, I have collected data in Robosuite. When I try to extract low dimensional observations, I get the following error

Error :
`ValueError: Error: joint has `range` but not `limited`. set the autolimits="true" compiler option, specify `limited` explicitly ("true" or "false"), or remove the `range` attribute.
Object name = robot0_joint1, id = 0, line = 182, column = -1`

 It would be appreciated if you help with this.

amandlek, 2024-02-28:
Which version of robosuite are you using? And which version of mujoco are you on?

dhanushTDS, 2024-02-29:
robomimic : 0.3.0
robosuite : 1.4.1
MuJoCo : 2.3.2

amandlek, 2024-02-29:
I think you should try upgrading robosuite to latest master - I had pushed a fix to use autolimits (see [here](https://github.com/ARISE-Initiative/robosuite/commit/b9d8d3de5e3dfd1724f4a0e6555246c460407daa#diff-b6759c796c91a056048f0289be030a926b3e320f677cb48851697c4e4f487c30R3)). That should hopefully fix the issue.

As for randomizing gripper position, I don't think that is currently supported, but you should try raising an issue in the robosuite repo to double check.

closed 2024-02-29 by amandlek

Dhanushvarma, 2024-03-01:
Isn't this already the latest version of the robosuite?

amandlek, 2024-03-01:
The version string was bumped before the commit I linked - so you may not necessarily be on a commit after that one.

Dhanushvarma, 2024-03-02:
So, I did check, that I am on the latest version and the fix you mentioned is indeed implemented in the version I am using, still the issue is there

amandlek, 2024-03-07:
I believe you need to either (1) re-collect the demos with the new settings or (2) postprocess the model xml contained in the demonstrations to add the xml change from the commit I linked you before