ARISE-Initiative/robomimic #143: train bc_transformer

Nimingez, 2024-03-08:
I collect my wipe datase from robosuite

python dataset_states_to_obs.py --dataset /home/xxx/robomimic/demo.hdf5 --output_name image_dense.hdf5 --done_mode 2  --camera_names agentview robot0_eye_in_hand --camera_height 84 --camera_width 84
I want to train with bc_transformer

python train.py --config ../config/default_templates/bc_transformer.json --dataset /home/nmz/robomimic/image.hdf5 

RuntimeError: cannot reshape tensor of 0 elements into shape [-1, 0] because the unspecified dimension size -1 can be any value and is ambiguous

amandlek, 2024-03-10:
The default config trains on object states not image observations (see [here](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/config/default_templates/bc_transformer.json#L112)) - is this what you wanted?

As for the error, my best guess is that there is something funny about the "object" key for the Wipe task (populated [here](https://github.com/ARISE-Initiative/robomimic/blob/b4f2d173f8161a1bedf13b5f045f05b4d6d38d9e/robomimic/envs/env_robosuite.py#L228) during `dataset_states_to_obs.py`) so it might be worth checking the shapes going into this observation key. This might actually be a robosuite issue (with the Wipe task).

Let us know what you find and we can follow up.

BuiDacHien, 2024-03-10:
Hey sir, trong lúc tôi chạy đoạn code sau: 

python robomimic/scripts/train.py --config robomimic/exps/templates/bc.json --dataset datasets/lift/ph/low_dim_v141.hdf5 --debug

Tôi gặp lỗi sau: 

100%|##########| 3/3 [00:00<00:00,  8.87it/s]
Train Epoch 1
{
    "Cosine_Loss": 0.5587675174077352,
    "L1_Loss": 0.0957380086183548,
    "L2_Loss": 0.19192640483379364,
    "Loss": 0.19192640483379364,
    "Optimizer/policy0_lr": 0.0001,
    "Policy_Grad_Norms": 0.2235456732159946,
    "Time_Data_Loading": 0.00018465916315714518,
    "Time_Epoch": 0.005668433507283529,
    "Time_Log_Info": 1.4901161193847656e-05,
    "Time_Process_Batch": 0.00012256701787312826,
    "Time_Train_Batch": 0.005309998989105225
}
video writes to /tmp/tmp_trained_models/test/20240310193926/videos/Lift_epoch_1.mp4
rollout: env=Lift, horizon=10, use_goals=False, num_episodes=2
100%|##########| 2/2 [00:18<00:00,  9.12s/it]

Epoch 1 Rollouts took 9.12014651298523s (avg) with results:
Env: Lift
{
    "Horizon": 10.0,
    "Return": 0.0,
    "Success_Rate": 0.0,
    "Time_Episode": 0.3040048837661743,
    "time": 9.12014651298523
}
save checkpoint to /tmp/tmp_trained_models/test/20240310193926/models/model_epoch_1_Lift_success_0.0.pth

Epoch 1 Memory Usage: 1238 MB

100%|##########| 3/3 [00:00<00:00, 51.14it/s]
Train Epoch 2
{
    "Cosine_Loss": 0.5759696165720621,
    "L1_Loss": 0.0873916173974673,
    "L2_Loss": 0.1788101146618525,
    "Loss": 0.1788101146618525,
    "Optimizer/policy0_lr": 0.0001,
    "Policy_Grad_Norms": 0.022320906980591342,
    "Time_Data_Loading": 0.0001711289087931315,
    "Time_Epoch": 0.0009937047958374023,
    "Time_Log_Info": 1.8628438313802085e-05,
    "Time_Process_Batch": 1.510779062906901e-05,
    "Time_Train_Batch": 0.0007715781529744466
}
video writes to /tmp/tmp_trained_models/test/20240310193926/videos/Lift_epoch_2.mp4
rollout: env=Lift, horizon=10, use_goals=False, num_episodes=2
100%|##########| 2/2 [00:12<00:00,  6.23s/it]

Epoch 2 Rollouts took 6.228485345840454s (avg) with results:
Env: Lift
{
    "Horizon": 10.0,
    "Return": 0.0,
    "Success_Rate": 0.0,
    "Time_Episode": 0.2076161781946818,
    "time": 6.228485345840454
}

Epoch 2 Memory Usage: 1256 MB

finished run successfully!
Exception ignored in: <function MjRenderContext.__del__ at 0x7f027bf3d310>
Traceback (most recent call last):
  File "/dis/camvan/BDH/robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
    self.gl_ctx.free()
  File "/dis/camvan/BDH/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "/home/camvan/anaconda3/envs/robomimic_venv/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
        err = EGL_NOT_INITIALIZED,
        baseOperation = eglMakeCurrent,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7f022bcc19c0>,
                <OpenGL._opaque.EGLSurface_pointer object at 0x7f028456a040>,
                <OpenGL._opaque.EGLSurface_pointer object at 0x7f028456a040>,
                <OpenGL._opaque.EGLContext_pointer object at 0x7f0284661f40>,
        ),
        result = 0
)
Exception ignored in: <function EGLGLContext.__del__ at 0x7f027bf3d160>
Traceback (most recent call last):
  File "/dis/camvan/BDH/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
  File "/dis/camvan/BDH/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "/home/camvan/anaconda3/envs/robomimic_venv/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
        err = EGL_NOT_INITIALIZED,
        baseOperation = eglMakeCurrent,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7f022bcc19c0>,
                <OpenGL._opaque.EGLSurface_pointer object at 0x7f028456a040>,
                <OpenGL._opaque.EGLSurface_pointer object at 0x7f028456a040>,
                <OpenGL._opaque.EGLContext_pointer object at 0x7f0284661f40>,
        ),
        result = 0
)

closed 2024-03-26 by amandlek