ARISE-Initiative/robomimic #148: How to transform actions to local robot end effector frame?

TongZhangTHU, 2024-03-26:
Hi,

Thanks for your great work!  I've observed that the actions in the data utilize the "OSC_POSE" controller type, indicating they are defined in the global coordinate frame. However, I'm currently exploring the possibility of obtaining actions relative to the local frame of the robot's end effector, (which might align more closely with the "IK_POSE" controller type, see [https://github.com/ARISE-Initiative/robosuite/blob/48c1b8a6c077d04399a00db05694d7f9f876ffc9/docs/demos.md?plain=1#L37](https://github.com/ARISE-Initiative/robosuite/blob/48c1b8a6c077d04399a00db05694d7f9f876ffc9/docs/demos.md?plain=1#L37) ）

I understand that I could potentially implement the coordinate transformation myself, but for the sake of accuracy and convenience, I'm curious if there are any existing methods or tools available to achieve this.

Thank you very much in advance!

amandlek, 2024-03-26:
I am not aware of an existing implementation of this for OSC in robosuite, but it should be straightforward to do this conversion.

closed 2024-03-26 by amandlek

sarthakdas, 2024-07-11:
Hi, did you get anything working for this? would be really helpful if you could share