ARISE-Initiative/robomimic #154: Question about an error when "Extracting Observations from MuJoCo states"

pengzhi1998, 2024-05-14:
Hi, thank you so much for providing this great repo for robot manipulation research!

However, I faced a problem when trying to extract image data from `demo_v141.hdf5` using: 
```
python dataset_states_to_obs.py --done_mode 2 \                               
--dataset ~/anaconda3/envs/multitask38/lib/python3.8/site-packages/datasets/square/ph/demo_v141.hdf5 \
--output_name image.hdf5 --camera_names agentview robot0_eye_in_hand --camera_height 84 --camera_width 84
```
while the `demo_v141` was downloaded via:
```
python download_datasets.py --tasks square --dataset_types ph --hdf5_types raw
```
The problem shows:
```
ROBOMIMIC WARNING(
    No private macro file found!
    It is recommended to use a private macro file
    To setup, run: python /home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robomimic/scripts/setup_macros.py
)
[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python /home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robosuite/scripts/setup_macros.py (macros.py:55)

============= Initialized Observation Utils with Obs Spec =============

using obs modality: low_dim with keys: []
using obs modality: rgb with keys: ['robot0_eye_in_hand_image', 'agentview_image']
==== Using environment with the following metadata ====
{
    "env_name": "NutAssemblySquare",
    "env_version": "1.4.1",
    "type": 1,
    "env_kwargs": {
        "has_renderer": false,
        "has_offscreen_renderer": true,
        "ignore_done": true,
        "use_object_obs": true,
        "use_camera_obs": true,
        "control_freq": 20,
        "controller_configs": {
            "type": "OSC_POSE",
            "input_max": 1,
            "input_min": -1,
            "output_max": [
                0.05,
                0.05,
                0.05,
                0.5,
                0.5,
                0.5
            ],
            "output_min": [
                -0.05,
                -0.05,
                -0.05,
                -0.5,
                -0.5,
                -0.5
            ],
            "kp": 150,
            "damping": 1,
            "impedance_mode": "fixed",
            "kp_limits": [
                0,
                300
            ],
            "damping_limits": [
                0,
                10
            ],
            "position_limits": null,
            "orientation_limits": null,
            "uncouple_pos_ori": true,
            "control_delta": true,
            "interpolation": null,
            "ramp_ratio": 0.2
        },
        "robots": [
            "Panda"
        ],
        "camera_depths": false,
        "camera_heights": 84,
        "camera_widths": 84,
        "reward_shaping": false,
        "camera_names": [
            "agentview",
            "robot0_eye_in_hand"
        ],
        "render_gpu_device_id": 0
    }
}

input file: /home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/datasets/square/ph/demo_v141.hdf5
output file: /home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/datasets/square/ph/image.hdf5
ep 0: wrote 127 transitions to group demo_0
ep 1: wrote 123 transitions to group demo_1
ep 2: wrote 124 transitions to group demo_2
ep 3: wrote 149 transitions to group demo_3
ep 4: wrote 154 transitions to group demo_4
ep 5: wrote 144 transitions to group demo_5
Traceback (most recent call last):
  File "dataset_states_to_obs.py", line 319, in <module>
  File "dataset_states_to_obs.py", line 189, in dataset_states_to_obs
  File "dataset_states_to_obs.py", line 78, in extract_trajectory
  File "/home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robomimic/envs/env_robosuite.py", line 132, in reset_to
  File "/home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robomimic/envs/env_robosuite.py", line 114, in reset
  File "/home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robosuite/environments/base.py", line 252, in reset
  File "/home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robosuite/environments/base.py", line 230, in _initialize_sim
  File "/home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robosuite/utils/binding_utils.py", line 1072, in from_xml_string
ValueError: Error: resource not found via provider or OS filesystem: '/home/pengzhi/anaconda3/envs/multitask38/lib/python3.8/site-packages/robosuite/models/assets/robots/panda/obj_meshes/link7_vis/link7_vis_7.obj'
```
I didn't find a solution to this. May I have your suggestions and guidance? Thanks a lot in advance!

yubinwang11, 2024-05-24:
facing the same questions

akansha-kalra, 2024-05-25:
Running into the same error as well!

yubinwang11, 2024-05-26:
The error can be solved by degrading mujoco from 3.1.5 to 3.1.4 according to the solutions in [https://github.com/ARISE-Initiative/robosuite/issues/481](url). Hope this can help you.

akansha-kalra, 2024-05-26:
> The error can be solved by degrading mujoco from 3.1.5 to 3.1.4 according to the solutions in [https://github.com/ARISE-Initiative/robosuite/issues/481](url). Hope this can help you.

Thank you so much!

closed 2024-05-30 by pengzhi1998

fancangning, 2024-06-02:
> The error can be solved by degrading mujoco from 3.1.5 to 3.1.4 according to the solutions in [https://github.com/ARISE-Initiative/robosuite/issues/481](url). Hope this can help you.

Thank you so much！！！！