ARISE-Initiative/robomimic #159: module 'robosuite' has no attribute 'make'

luccachiang, 2024-06-11:
Hi, I'm using the `robocasa` branch, but got the AttributeError in this line:
https://github.com/ARISE-Initiative/robomimic/blob/2e32a25f9aea6ba486de39e1d8d353244fe8a039/robomimic/envs/env_robosuite.py#L94

Is there something wrong with this branch?

snasiriany, 2024-06-11:
Hi @luccachiang can you please print the full stack trace, and let me know all the steps to reproduce the issue?

luccachiang, 2024-06-12:
Thanks for your prompt response. @snasiriany I believe the following steps are the minimum to reproduce the error.

1. Install following https://github.com/robocasa/robocasa#installation. Note that insterestingly after installing `robosuite (step 3)`, I do not have the `make` attribute error. However, after installing `robocasa (step 4)`, I cannot import 'make'.
[image]

2. Install following https://github.com/ARISE-Initiative/robomimic/tree/robocasa#installation
3. Use CLI python in this conda env to test robosuite
```
from robosuite import make
```

Based on the steps above, I can solve this problem by re-executing robosuite installation with `pip install -e .`. However, if it is convenient for you, please check why this problem occurs:)

luccachiang, 2024-06-12:
Sorry, I've finally found the solution. I encountered this problem because I git clone robosuite in the folder of robocasa. As a result, the folder name `robosuite` has conflicts with the package name. After rename the folder name, I solved this problem :)

closed 2024-06-12 by luccachiang