ARISE-Initiative/robomimic #161: PnPCounterToMicrowave h5py file error

fortyMiles, 2024-06-14:
Hi, thanks for your contribution on this environment. 

Recently, I try to install and run the training examples. I met an error " /home/username/robocasa/datasets/v0.1/single_stage/kitchen_pnp/PnPCounterToMicrowave/2024-04-27/demo_gentex_im128_randcams.hdf5
run failed with error:
Unable to open file (file signature not found)" 

with   

File "h5py/h5f.pyx", line 102, in h5py.h5f.open
OSError: Unable to open file (file signature not found)
--- 

I have tried to re-download and overwrite several times of **PnPCounterToMicrowave** data. But this issue still exists. 

Can you help me to solve that? 

Best Regards! 

[image]

joeljang, 2024-06-14:
I get the same error as well.

snasiriany, 2024-06-16:
Hi all! Just fixed, and addressed here: https://github.com/robocasa/robocasa/issues/20

closed 2024-06-16 by snasiriany