ARISE-Initiative/robomimic #163: ImportError

Alpslee, 2024-06-17:
try to run:
python robomimic/scripts/config_gen/bc_xfmr.py --name <experiment-name>
while got the error:
 ImportError: cannot import name 'rotate_2d_point' from 'robosuite.utils.transform_utils'

it seems the 'bc_xfmr.py' file name is wrong, and I corrected  it then run:
python robomimic/scripts/config_gen/bc_xfmr_gen.py --name PrepareCoffee

Traceback (most recent call last):
File "/Users/randli/robomimic/robomimic/scripts/config_gen/bc_xfmr_gen.py", line 109, in <module>
make_generator(args, make_generator_helper)
File "/Users/randli/robomimic/robomimic/scripts/config_gen/helper.py", line 529, in make_generator
generator = make_generator_helper(args)
File "/Users/randli/robomimic/robomimic/scripts/config_gen/bc_xfmr_gen.py", line 23, in make_generator_helper
(get_ds_cfg("single_stage", src="human", eval=EVAL_TASKS, filter_key="50_demos"), "human-50"),
File "/Users/randli/robomimic/robomimic/scripts/config_gen/helper.py", line 580, in get_ds_cfg
from robocasa.utils.dataset_registry import get_ds_path, SINGLE_STAGE_TASK_DATASETS, MULTI_STAGE_TASK_DATASETS
File "/Users/randli/robocasa/robocasa/init.py", line 6, in <module>
from robocasa.environments.kitchen.kitchen import (
File "/Users/randli/robocasa/robocasa/environments/kitchen/kitchen.py", line 18, in <module>
from robocasa.utils.placement_samplers import SequentialCompositeSampler, UniformRandomSampler
File "/Users/randli/robocasa/robocasa/utils/placement_samplers.py", line 9, in <module>
from robosuite.utils.transform_utils import quat_multiply, euler2mat, mat2quat, convert_quat, rotate_2d_point
ImportError: cannot import name 'rotate_2d_point' from 'robosuite.utils.transform_utils' (/Users/randli/robosuite/robosuite/utils/transform_utils.py)

snasiriany, 2024-06-17:
The function should exist here: https://github.com/ARISE-Initiative/robosuite/blob/29ee7efb49476d5bcc5bb43cf75105903eadb1a7/robosuite/utils/transform_utils.py#L932

Are you using the `robocasa_v0.1` branch of robosuite? You may be on a different branch

Alpslee, 2024-06-18:
> The function should exist here: https://github.com/ARISE-Initiative/robosuite/blob/29ee7efb49476d5bcc5bb43cf75105903eadb1a7/robosuite/utils/transform_utils.py#L932
> 
> Are you using the `robocasa_v0.1` branch of robosuite? You may be on a different branch

Thanks. yes,i am using the robocall_v0.1 branch of robosuite. I added the function in my robosuite.utils.transform_utils, try to run it again on macbook, got new error:
(robocasa) randli@Rands-Laptop robomimic %  python robomimic/scripts/config_gen/bc_xfmr_gen.py --name PrepareCoffee
WARNING: mimicgen environments not imported since mimicgen is not installed!
Num exps: 1

python /Users/randli/robomimic/robomimic/scripts/train.py --config /Users/randli/tmp/autogen_configs/ril/bc/robocasa/im/06-18-PrepareCoffee/06-18-24-16-59-15/json/seed_123_ds_human-50.json

amandlek, 2024-07-01:
Are you talking about the "WARNING"? You should be able to ignore that safely.

closed 2024-07-01 by amandlek