ARISE-Initiative/robomimic #170: Error when loading dataset as env

niks64, 2024-07-01:
Code:
```
dataset = "../robomimic/datasets/lift/ph/low_dim_v141.hdf5"
    env_meta = FileUtils.get_env_metadata_from_dataset(dataset)

    env = EnvUtils.create_env_for_data_processing(
        env_meta=env_meta,
        camera_names=[], 
        camera_height=84, 
        camera_width=84,
        reward_shaping=False,
    )
```

Error:
```
File "/anaconda3/envs/robomimic_env/lib/python3.12/site-packages/robosuite/robots/robot.py", line 148, in reset
 self.base_ori = T.mat2quat(self.sim.data.get_body_xmat(self.robot_model.root_body).reshape((3, 3)))

TypeError: can't unbox array from PyObject into native value.  The object maybe of a different type
```

Getting this error when loading one of the robomimic datasets. I am on a M1 Mac. My numba version is 0.60.0. Any idea what could be causing this error?

amandlek, 2024-07-01:
I'm not too sure. Perhaps you could check the types of each object to understand what could be going on?

niks64, 2024-07-02:
The error is thrown when doing T.mat2quat(). The data being passed into the function is <class 'numpy.ndarray'>, and Shape: (3, 3). None of the lines inside the mat2quat function execute but it has a @jit_decorator. 

I turned NUMBA off in macros and there is no longer an error.

closed 2024-07-02 by niks64