ARISE-Initiative/robomimic #179: Video cannot be opened

noooob-coder, 2024-07-23:
When I run p`ython robomimic/scripts/train.py --config robomimic/exps/templates/bc.json --dataset datasets/lift/ph/low_dim_v141.hdf5 --debug` the following error occurs. Program cannot run properly. When I delete `--debug`, although it can run normally, the video in `robomimic/bc_trained_models/test/20240723101813/videos `cannot be opened normally.

```
Exception ignored in: <function MjRenderContext.__del__ at 0x7f5dbe1c1dc0>
Traceback (most recent call last):
  File "/home/user/anaconda3/envs/robomimic_venv/lib/python3.8/site-packages/robosuite/utils/binding_utils.py", line 199, in __del__
    self.gl_ctx.free()
  File "/home/user/anaconda3/envs/robomimic_venv/lib/python3.8/site-packages/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "src/errorchecker.pyx", line 58, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglMakeCurrent,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f5dbb869f40>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f5dbe544bc0>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f5dbe544bc0>,
		<OpenGL._opaque.EGLContext_pointer object at 0x7f5dbe544340>,
	),
	result = 0
)
Exception ignored in: <function EGLGLContext.__del__ at 0x7f5dbe1c1c10>
Traceback (most recent call last):
  File "/home/user/anaconda3/envs/robomimic_venv/lib/python3.8/site-packages/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
  File "/home/user/anaconda3/envs/robomimic_venv/lib/python3.8/site-packages/robosuite/renderers/context/egl_context.py", line 149, in free
    EGL.eglMakeCurrent(EGL_DISPLAY, EGL.EGL_NO_SURFACE, EGL.EGL_NO_SURFACE, EGL.EGL_NO_CONTEXT)
  File "src/errorchecker.pyx", line 58, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglMakeCurrent,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f5dbb869f40>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f5dbe544bc0>,
		<OpenGL._opaque.EGLSurface_pointer object at 0x7f5dbe544bc0>,
		<OpenGL._opaque.EGLContext_pointer object at 0x7f5dbe544340>,
	),
	result = 0
)

```

amandlek, 2024-07-23:
This error is normal and expected and can safely be ignored. I'm guessing the run with the debug flag ran normally, and the run without the debug flag didn't finish enough rollouts during evaluation so the video was malformed.

closed 2024-07-23 by amandlek

amandlek, 2024-07-24:
This is precisely what the `--debug` flag is supposed to do - test a training run quickly with 2 epochs of training. To train for longer, simply omit the `--debug` flag.