ARISE-Initiative/robomimic #183: how to configure discrete action in bc.json

gemcollector, 2024-07-27:
hi, great work! I wanna ask how should I modify the `bc.json` to adapt to discrete action output?

amandlek, 2024-07-27:
we do not support discrete actions currently - that would require extra implementation (for example, a separate policy network that treats action outputs as classification, and some postprocessing logic to take existing continuous actions in demonstrations and relabel them into discrete actions)

closed 2024-07-27 by amandlek

arhanjain, 2024-09-01:
@amandlek if i were to implement the new algo/policy to support discretized actions for already discretized data (many including myself use robomimic as training infra so we already have discrete data from other sources), would it be possible to have it merged?

amandlek, 2024-09-05:
Yes, definitely! It would be great to have a PR for this.