ARISE-Initiative/robomimic #185: mj_loadXML error: b"XML Error: Schema violation: unrecognized attribute: 'autolimits'\n\nElement 'compiler'

btbuyccycc, 2024-08-07:
I got this error when I was running extract _ OBS _ from _ raw _ datasets.sh. After checking, my mujoco_py version is 2.1.2.14. I want to know the correct version of mujoco_py when running this document.

amandlek, 2024-08-14:
My best guess is that you need to upgrade your mujoco version (e.g. `pip install -U mujoco`). Depending on your version of robosuite, you shouldn't be using mujoco_py at all

closed 2024-08-14 by amandlek

Ding-Kaiyue, 2025-01-10:
I have the same question.

jangyicheng, 2025-05-19:
Hey, I also encountered almost the same issue today. Here are some possible solutions:  

Your problem seems to be caused by the newer version of MuJoCo removing certain attributes (such as `autolimits`), while Robosuite or some older models might still rely on them, leading to parsing failures.  

Below are two possible solutions I tried:  

**Solution 1: Directly modify the XML file**  
Go to the file where the error occurs:  
```bash  
vim /root/miniconda3/envs/data/lib/python3.9/site-packages/robosuite/environments/base.py  
```  
Then modify the `xml_string` like this:  
```python  
if xml_string != None:  
    xml_string = xml_string.replace('autolimits="true"', '')  
    xml_string = xml_string.replace('autolimits="false"', '')  
    import re  
    xml_string = re.sub(r'gravcomp="[^"]*"', '', xml_string)  
    ...  
    self.mjpy_model = load_model_from_xml(xml_string) if xml_string else self.model.get_model(mode="mujoco_py")  
```  
However, it seems many attributes are incompatible, so I eventually gave up on this approach.  

**Solution 2: Upgrade Robosuite (recommended)**  
The best solution is likely to upgrade Robosuite:  
```bash  
pip install --upgrade robosuite  
```  
Running this command resolved my issue—hope it works for you too!