ARISE-Initiative/robomimic #189: Supporting custom backbone networks for VisualCore

arhanjain, 2024-08-19:
Because of the way the VisualCore class is written, it only allows backbones to be from the supplied base nets in the default repository. Would it make sense to instead use a registry where inheriting from ConvBase would allow any custom network to be used as a backbone for VisualCore?

`backbone_kwargs = extract_class_init_kwargs_from_dict(cls=eval(backbone_class), dic=backbone_kwargs, copy=True)`

Perhaps there's a better way to approach this than I'm thinking. My current workout is to define my own Core class that allows for using this sort of registry.

danfeiX, 2024-08-20:
Hi @arhanjain , thanks for the note. I think a register is a great idea. It should be simple to implement too in base_net.py. Would you like to create a PR for it?

amandlek, 2024-09-05:
Addressed by #190

closed 2024-09-05 by amandlek