ARISE-Initiative/robomimic #193: using the Robomimic dataset with the corresponding Robosuite environment.

baimukashev, 2024-09-07:
Hello, I had a question about using the Robomimic dataset with corresponding Robosuite env. 

Script that extracts obs from raw data <robomimic/scripts/dataset_states_to_obs.py> creates env, and its obsersations consist of:
dict_keys(['object', 'robot0_joint_pos', 'robot0_joint_pos_cos', 'robot0_joint_pos_sin', 'robot0_joint_vel', 'robot0_eef_pos', 'robot0_eef_quat', 'robot0_eef_vel_lin', 'robot0_eef_vel_ang', 'robot0_gripper_qpos', 'robot0_gripper_qvel'])

While Robosuite for the same task and robot creates the env with observations:
odict_keys(['robot0_joint_pos_cos', 'robot0_joint_pos_sin', 'robot0_joint_vel', 'robot0_eef_pos', 'robot0_eef_quat', 'robot0_proprio-state']) . The last 'robot0_proprio-state' is used when wrapped in gym interface. How could I extract the same observations from dataset so that it mathches 'robot0_proprio-state' (both keys and order). Thanks

Versions installed.
robomimic                 0.3.0                     dev_0    <develop>
robosuite                 1.4.1                    pypi_0    pypi

baimukashev, 2024-09-07:
In 'environment.md' I found the following. To match the number, I believe, robot0_eef_quat  should also be included. But still I could not find in which order they are concatenated. 

- `robot0_proprio-state`, `robot1_proprio-state` : proprioception observations for each robot arm. This includes the arm joint positions (encoded using `sin` and `cos`), arm joint velocities, end effector pose, gripper finger positions, and gripper finger velocities. The shape for this modality is flat (e.g. `(N,)`).

amandlek, 2024-10-06:
You can see the logic for how robosuite observations are filtered and re-named [here](https://github.com/ARISE-Initiative/robomimic/blob/29d6ca229dec3327f87b54cf1688a94177f92af6/robomimic/envs/env_robosuite.py#L237) and modify it for your own use case.

closed 2024-10-06 by amandlek