ARISE-Initiative/robomimic #196: Question about causal masking in BC_transformer

kang-jaehyun, 2024-09-27:
Hello, thank you for maintaining such an excellent dataset and environment for robotics research. I am currently training behavior cloning using robomimic in the robocasa environment and have a question regarding the BC_transformer that is implemented as a baseline.

1. When predict_future_acs is set to true, it seems like the model takes inputs from o_t-10 to o_t and predicts actions from a_t to a_t+10. However, the implementation uses causal masking in attention. Since it’s predicting future actions, it doesn’t seem necessary to use causal masking. Could you explain the reasoning behind this implementation?

2. Also, as far as I understood, features corresponding to o_t-10 to o_t go through attention and are supervised to predict a_t to a_t+10. Does this mean that, in the end, o_t-10 predicts a_t, and o_t-9 predicts a_t+1 and so on?

snasiriany, 2024-09-29:
Thanks for raising this issue! It seems like we have a flaw in our code. We intended the network to be non-causal. And in fact in the configs we try to disable causal self-attention: https://github.com/ARISE-Initiative/robomimic/blob/1651ee2ea4e149cd8cffbeb77323ed221fa1d9a7/robomimic/exps/templates/bc_transformer.json#L103

However, the causal config flag isn't reflected in the policy codebase at all. We this previously in a previous version of the codebase but it seems in the process we removed it. I'll update the codebase to fix this.

snasiriany, 2024-09-29:
Fixed with this commit: https://github.com/ARISE-Initiative/robomimic/commit/4e693812ad13a308beec97ae08364dbdf53fc930
Please try again and let me know if there are any issues or have any questions.

closed 2024-09-29 by snasiriany