ARISE-Initiative/robomimic #197: Run test.sh with error bc-rgb-{}: failed with error: 'ResNet18Conv'

Chezacar, 2024-09-29:
Thank you for your help. When I run 
$ cd tests
$ bash test.sh
It will raise error like:
robomimic/robomimic/models/obs_core.py", line 102, in __init__
    cls = ObsUtils.OBS_ENCODER_CORES[backbone_class],
KeyError: 'ResNet18Conv'

bc-image-{xxxx}: failed with error:
'ResNet18Conv'

bc-rgb-{xxxx}: failed with error:
'ResNet18Conv'

How to resolve this issue? Is this because the version mismatch? I found that there is no code for class 'ResNet18Conv' to run register_encoder_core function at robomimic/utils/obs_utils.py.

KAI-AVG, 2024-10-07:
robomimic.models.obs_core.py line 102中 backbone_kwargs中cls换成了Resnet18Conv，同时在该文件中导入了robomimic.models.base_nets导入了ResNet18Conv

Chezacar, 2024-10-07:
In fact, when I use the commit from before July 15th, I don’t encounter the KeyError: ‘ResNet18Conv’. I suspect there is some mismatch in the latest commit​.

KAI-AVG, 2024-10-07:
> In fact, when I use the commit from before July 15th, I don’t encounter the KeyError: ‘ResNet18Conv’. I suspect there is some mismatch in the latest commit​.

Yes, I also think the latest release has some minor issues. I just checked the source code of version 0.2.0 and there is no "obs_core.py" file. I suspect that the author rewrote the "model" module in the latest version.

danfeiX, 2024-10-07:
Thanks for bringing this up. Here is the fix #199 . Will merge ASAP after reviewing.

closed 2024-10-08 by danfeiX