ARISE-Initiative/robomimic #198: Diffusion Policy

arhanjain, 2024-10-06:
Wondering if there's a reason diffusion policy hasn't been merged into master? 

There's also a few different branches related to diffusion so it's a bit confusing which one is stable & up to date.

amandlek, 2024-10-06:
Great question. The other branches offer a preliminary diffusion policy implementation but we would like to clean it up and also run some tests to make sure the implementation is on par with the "official" diffusion policy codebase.

sarthakdas, 2025-02-11:
which branch is currently working? which version of diffusers do you have getting this error: 

RuntimeError: Failed to import diffusers.models.unets.unet_2d_condition because of the following error (look up to see its traceback):
module 'torch' has no attribute 'float8_e4m3fn'

amandlek, 2025-02-11:
You can find more info about the `robocasa` branch and using it [here](https://robocasa.ai/docs/use_cases/policy_learning.html)

amandlek, 2025-03-29:
robocasa uses a different version of the robomimic codebase for policy training (the `robocasa` branch of robomimic) - my link refers to files on that branch of robomimic. 

As for a more thorough evaluation of diffusion policy, we are working on standardizing and cleaning up the implementation and it will be made available on our `master` branch in the next robomimic version. For now, you could use the implementation on the `robocasa` branch, which is more of an "experimental" branch of the code

Hope that's helpful!

amandlek, 2025-07-08:
Closing since robomimic v0.5 implements diffusion policy.

closed 2025-07-08 by amandlek