ARISE-Initiative/robomimic #206: AssertionError: OSC_POSE controller is specified, but not imported or loaded

AbdElrahmanMostafaRifaat1432, 2024-11-30:
[image]

AbdElrahmanMostafaRifaat1432, 2024-11-30:
Also your Colab code provide the same error

lonely2silence, 2024-12-02:
Yes, I also have a same problem.

lonely2silence, 2024-12-02:
It's a little confusing when I need to choose my keyboard layout.

amandlek, 2024-12-02:
What's your robosuite version? I'd recommend using robosuite v1.4 or lower for now, as robosuite v1.5 has not been tested thoroughly with the examples in robomimic.

AbdElrahmanMostafaRifaat1432, 2024-12-04:
is there an intention to put diffusion policy in robomimic or how to integerate it if you could help me I will appreciate that

amandlek, 2024-12-04:
We have not officially integrated it (requires further cleanup and testing) but there is an unofficial release on the `robocasa` branch: https://github.com/ARISE-Initiative/robomimic/tree/robocasa

closed 2024-12-04 by amandlek

krishanrana, 2024-12-13:
> What's your robosuite version? I'd recommend using robosuite v1.4 or lower for now, as robosuite v1.5 has not been tested thoroughly with the examples in robomimic.

But not all environments are available in v1.4. Please can you fix this.