ARISE-Initiative/robomimic #210: Paired dataset vs PH dataset

rl-max, 2024-12-19:
Hi, thank you for the great works!

I have one question regarding the dataset.

Does Can-paired dataset contain Can-PH (proficient human)? To be precise, Are good trajectories in Can-paired dataset exactly same as ones in Can-PH?

amandlek, 2024-12-19:
It has been a while since we collected those datasets so I wasn't 100% sure, but I'm confident that they are not the same. One easy way to verify this is to take one "good" trajectory in Can-paired, check its length, and find all trajectories in Can-PH with the same length. Next, check the states and check for exact equality with the subset of trajectories in Can-PH. Hopefully that makes sense!

Here's a simple script I wrote to validate this. When running it, I didn't find an exact match with any trajectories.

```python
"""
Helper script to validate whether the good trajectories in Can-Paired are a subset of the ones in Can-PH.
"""
import os
import h5py
import numpy as np

# path to your datasets
CAN_PAIRED_PATH = os.path.expanduser("~/Downloads/can_paired.hdf5")
CAN_PH_PATH = os.path.expanduser("~/Downloads/can_ph.hdf5")

if __name__ == "__main__":

    # open files
    f_paired = h5py.File(CAN_PAIRED_PATH, "r")
    f_ph = h5py.File(CAN_PH_PATH, "r")

    # find a demo that is successful in can paired dataset
    paired_succ_demo_key = None
    for demo_key in f_paired["data"].keys():
        if (np.sum(f_paired["data/{}/rewards".format(demo_key)]) > 0.):
            paired_succ_demo_key = demo_key
            break

    assert paired_succ_demo_key is not None
    paired_succ_ep_grp = f_paired["data/{}".format(paired_succ_demo_key)]
    paired_succ_ep_length = paired_succ_ep_grp["actions"].shape[0]
    print("got successful Can-Paired demo key {} with length {}".format(paired_succ_demo_key, paired_succ_ep_length))

    # find all demos in can ph with the same length
    ph_same_length_demo_keys = []
    for demo_key in f_ph["data"].keys():
        if f_ph["data/{}/actions".format(demo_key)].shape[0] == paired_succ_ep_length:
            ph_same_length_demo_keys.append(demo_key)

    print("got ph demo keys {} with same episode length".format(ph_same_length_demo_keys))

    # check for exact state equality in each demo
    paired_succ_ep_states = paired_succ_ep_grp["states"][()]
    state_errors = []
    for demo_key in ph_same_length_demo_keys:
        ep_grp = f_ph["data/{}".format(demo_key)]
        ep_states = ep_grp["states"][()]
        if np.all(np.equal(ep_states, paired_succ_ep_states)):
            print("got ph demo key {} with same states".format(demo_key))
        state_errors.append(np.linalg.norm(ep_states - paired_succ_ep_states))

    print("got state errors: {}".format(state_errors))

    # close files
    f_paired.close()
    f_ph.close()
```

closed 2024-12-19 by amandlek

rl-max, 2024-12-20:
Thank you for the answer and code. It helped !!