ARISE-Initiative/robomimic #211: PNG size must be integer multiple of gridsize in texture

CallMeFrozenBanana, 2024-12-26:
I change the version of robosuit from 1.5.0 to 1.4.0 because lack of the "link0_vis.mtl" file in the path "robosuite/models/assets/robots/panda/meshes"
But now I encountering a new problems ⬇️
Error infor：
'''
Error: PNG size must be integer multiple of gridsize in texture '/data/yenianjin/env/envs/lerobot/lib/python3.10/site-packages/robosuite-1.4.0-py3.10.egg/robosuite/models/assets/objects/../textures/cereal.png' (id 8)
Object name = Cereal_tex-cereal, id = 8, line = 18
  File "/data/yenianjin/project/robosuit/robomimic/robomimic/scripts/get_dataset_info.py", line 62, in playback_trajectory
    env.reset_to(initial_state_dict)
  File "/data/yenianjin/project/robosuit/robomimic/robomimic/scripts/get_dataset_info.py", line 164, in <module>
    playback_trajectory(ep, f, video_writer, env)
ValueError: Error: PNG size must be integer multiple of gridsize in texture '/data/yenianjin/env/envs/lerobot/lib/python3.10/site-packages/robosuite-1.4.0-py3.10.egg/robosuite/models/assets/objects/../textures/cereal.png' (id 8)
Object name = Cereal_tex-cereal, id = 8, line = 18
'''

Why ？ T T
Any suggestion is appreciate！

CallMeFrozenBanana, 2024-12-27:
I find that robomimic works well in the `offline_study` branch of the robosuite. And if the python version is greater than 3.10, we need to change the `from collections import Iterable` to `from collections.abc import Iterable`, `collections.Iterable` to `collections.abc.Iterable` in the source scripts of robosuite

closed 2024-12-27 by CallMeFrozenBanana

amandlek, 2024-12-28:
This is strange - but we've actually only tried robosuite v1.4.1 not v1.4, so I wonder if the issue stems from that version difference. [Here](https://robomimic.github.io/docs/datasets/robomimic_v0.1.html#downloading) is a note on the robosuite version to use.