ARISE-Initiative/robomimic #212: Colab quickstart run all error, section 6 first cell

francis-bilodeau, 2025-01-08:
Got this error while doing a run all of the colab tutorial robomimic_get_started.ipynb.

```
AssertionError                            Traceback (most recent call last)
<ipython-input-12-6f92859389ef> in <cell line: 7>()
      5 env_meta = FileUtils.get_env_metadata_from_dataset(dataset_path)
      6 
----> 7 env = EnvUtils.create_env_from_metadata(
      8     env_meta=env_meta,
      9     env_name=env_meta["env_name"],

13 frames
/usr/local/lib/python3.10/dist-packages/robosuite/controllers/composite/__init__.py in composite_controller_factory(type, sim, robot_model, grippers)
      6 
      7 def composite_controller_factory(type, sim, robot_model, grippers):
----> 8     assert type in REGISTERED_COMPOSITE_CONTROLLERS_DICT, f"{type} controller is specified, but not imported or loaded"
      9     # Note: Currently we assume that the init arguments are same for all composite controllers. The situation might change given new controllers in the future, and we will adjust accodingly.
     10 

AssertionError: OSC_POSE controller is specified, but not imported or loaded
```

The error originate cell of the "6. Evaluate and visualize trained policy" section.

amandlek, 2025-01-08:
This is likely an issue with the robosuite version - please use version v1.4.1 or the `offline_study` branch instead of v1.5 (latest master) for now.

cc @danfeiX for vis

danfeiX, 2025-01-10:
Let us know if changing version resolves the issue @francis-bilodeau

carloskiyoto, 2025-01-10:
> Let us know if changing version resolves the issue @francis-bilodeau

I had the same problem, I changed to v1.4.1 and worked! Thank you @amandlek

francis-bilodeau, 2025-01-10:
> This is likely an issue with the robosuite version - please use version v1.4.1 or the `offline_study` branch instead of v1.5 (latest master) for now.
> 
> cc @danfeiX for vis

This worked, thank you!

```
#install robosuite
!pip install robosuite==1.4.1
```

closed 2025-01-10 by francis-bilodeau