ARISE-Initiative/robomimic #213: rollout takes so much time

dzy666fly, 2025-01-13:
Hi, thank u so much for sharing this wonderful work!

I have two workstations with the same system (Ubuntu 20.04), Nvidia driver (470.256.02 ), GPU (RTX 3090), robomimc version (v0.2.1), and robosuite version (offline_study), but the rollout time of runing BC-RNN for the same task is different.

This is good.
```
# transport
# train
100%|##########| 400/400 [01:15<00:00,  5.29it/s]
# validate
100%|##########| 50/50 [00:10<00:00,  4.85it/s]
# rollout
100%|##########| 50/50 [1:00:58<00:00, 73.17s/it]

# square
# train
100%|##########| 400/400 [00:32<00:00, 12.14it/s]
# validate
100%|##########| 50/50 [00:05<00:00,  9.47it/s]
# rollout
100%|##########| 50/50 [10:20<00:00, 12.41s/it]
```

And this is worse; the simpler task `Square` needs one hour and the more complex task `Transport` needs over six hours.
```
# transport
# train
100%|##########| 400/400 [01:04<00:00,  6.21it/s]
# validate
100%|##########| 50/50 [00:10<00:00,  4.96it/s]
# rollout
38%|###8      | 19/50 [2:26:39<4:05:41, 475.53s/it]

# square
# train
100%|##########| 400/400 [00:35<00:00, 11.15it/s]
# validate
100%|##########| 50/50 [00:06<00:00,  8.25it/s]
# rollout 
94%|#########3| 47/50 [58:17<03:58, 79.67s/it]
```

So why did this happen? Even I reinstalled my system for the worse one, it did not change. How can I solve this problem?

Looking forward to ur reply.
Thank u so much.

amandlek, 2025-01-13:
my best guess is that one of your machines is not using the GPU for rendering images during rollouts - since you're using the `offline_study` branch of robosuite, you'll be using `mujoco-py` - I would verify whether mujoco-py is installed with the GPU build (and not the cpu build)

closed 2025-01-13 by amandlek

dzy666fly, 2025-01-14:
Thank u so much, after testing u r right :laughing: . I will try to install the proper `mujoco-py` version